wled/WLED #2817: WLED CODE FLOW

Gpadhiyar, 2022-10-07:
I am a beginner at this project structure. So, I do not understand the project structure or project workflow. I do not understand how to communicate applications to ESP32, how to send and receive data from an application to ESP32, and ESP32 to an application. I tried my best to understand the working process of WLED, but I didn't.

Please provide me with some workflow or guidance for understanding the working process of the project.

Aircoookie, 2022-10-07:
Hey, I'm not sure what you mean with project structure or workflow. If you are looking to make an application that communicates with the WLED ESP32, check the JSON API documentation: https://kno.wled.ge/interfaces/json-api/

blazoncek, 2022-10-07:
Other than what @Aircoookie said, join the Discord or [forum](https://wled.discourse.group) where friendly people will help you get started.

Gpadhiyar, 2022-10-08:
If i am mack coustom application then which perameter send to esp32

blazoncek, 2022-10-08:
Please use forum or Discord.

closed 2022-10-08 by blazoncek