wled/WLED #2819: add link to project page on install.wled.me

petaflot, 2022-10-08:
**Is your feature request related to a problem? Please describe.**
first time I was looking for wled (a friend had talked to me about it) my searches basically all returned install.wled.me.

since I did not have chrome or edge, I was simply faced with a frustrating error message that said "browser not supported"

**Describe the solution you'd like**

whether the browser is supported or not, install.wled.me should have a link to kno.wled.ge ; and possibly the present github repository.

ESPECIALLY if the browser is not supported, but really it should not depend on that

**Describe alternatives you've considered**

keeping my existing code for LED control

blazoncek, 2022-10-08:
Actually Google correctly points to WLED KB.
[image]

And the install instructions clearly state that Chrome or Edge are required. The other install methods are described as well.
If you are unable or unwilling to use Chrome or Edge (like I am) you still have the possibility to upload firmware to ESP with many other methods, none of them strictly tied to WLED as you can upload any firmware.
[image]

I do agree that having a link to KB on web installer would be beneficial.

Aircoookie, 2022-10-21:
Just added two links to github and KB to the installer :)

closed 2022-10-21 by Aircoookie