wled/WLED #2820: AEST Time Is Incorrect For DayLight Savings In October

greensleeves149, 2022-10-09:
### What happened?

On the first Sunday in October each year, the AEST time zone moves forward by one hour. Last Sunday (2nd October), the current time displayed in the WLED app was NOT updated to reflect the new Daylight savings time. (I saw this under Time and Macros>Time Setup>Current local Time.) I ended up requesting help in your Discord server and I had "blazoncek" kindly investigate and he came back to me with what we think is the reason why it didn't update on the 2nd October. 

In the attached file, I think the issue is that the first line of code with 
`tcrDaylight = {Second, Sun, Oct, 2, 660}  //AEDT = UTC + 11 Hours`

I'm guessing that the "Second" refers to the Second Sunday of October to start Day Light Savings Time?
Right now as I type this, it's Sunday 9th October (Second Sunday) and I've noticed that the "Current Local Time" is now CORRECT. However, for the last 7 days, it has been incorrect, meaning all my lights turn on and off at the Non Day Light Saving Time. 

So maybe the fix is:
``tcrDaylight = {First, Sun, Oct, 2, 660}  //AEDT = UTC + 11 Hours``

[image]

### To Reproduce Bug

'Current Local Time' in the WLED app in Australia between the first Sunday of October to the Second Sunday of October is incorrect as it fails to advance the time forward by one hour to reflect the new Day Light Savings time. 

### Expected Behavior

On the first Sunday in October each year, AEST time needs to advance forward by one hour. For example if the old time was 10AM, the WLED Current Local Time should show 11AM. (Daylight savings time ends in the first Sunday of April- which I think is correct in the code above. )

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

domdfcoding, 2022-10-12:
I think it's worth double checking the other timezones too. For instance NZ is down as second Sunday in September, but [according to the New Zealand government](https://www.govt.nz/browse/recreation-and-the-environment/daylight-saving/) it changes on the last Sunday.

https://github.com/Aircoookie/WLED/blob/e362b3b6aa524441c1ab963c4a470a207e83f922/wled00/ntp.cpp#L100-L104

Aircoookie, 2022-10-12:
Thank you for bringing this to my attention! Both NZ and AEST should now be fixed in 0_14 :)

closed 2022-10-12 by Aircoookie

greensleeves149, 2022-10-13:
Great. Thank you for actioning this so quickly too.

commit d6749d30 (2022-10-12): Fixed AEST and NZST DST change dates (resolves #2820 )