wled/WLED #2821: Perhaps a future release can support routers that have the 2.4ghz and 5.0ghz combined.

ShawMerlin, 2022-10-09:
A lot of the newer routers are coming with the 2.4ghz and 5.0ghz together.  I know this does not work with WLED as it needs them separated.

A future revision that would allow it to connect to networks that are combined as we are seeing more and more routers coming default this way.

We love WLED and are so appreciative of all the work you and your team puts into this!!

Maxcodesthings, 2022-10-10:
I have no trouble connecting to an SSID broadcasting a combined 2.4 and 5 Ghz. Your hardware might be limiting you

ShawMerlin, 2022-10-10:
It most likely is the hardware on the router but it's a new Verizon 5G Home Router and until we separated the bands, it would not connect.  When I was googling it, it seemed like a common issue so thought I would share it here.

I'm not saying it will impact everyone on a combined band, but might be worth looking into further.

blazoncek, 2022-10-10:
I have only one SSID on both frequencies (various channels) and every ESP on my network (including WLED devices) successfully connects.
This is not a problem of ESP firmware but the way router handles connections in such case.

Upgrade firmware on your router or replace it with a more compatible one.

softhack007, 2022-10-10:
Same for me, I have a router with 2.4ghz and 5ghz using the same network name. No problems.
Please keep in mind that ESP devices only have 2.4ghz on board, they cannot even "see" the 5ghz network.

I'r suggest to close this one, as WLED cannot do anything about 2.4ghz/5ghz co-existence. WLED simply does not know that there is a 5ghz WiFi network.

closed 2022-10-10 by ShawMerlin