wled/WLED #2822: receive data from android application in esp32.

govind183, 2022-10-10:
when user use android app  and select any effect or color etc.

at that moment the at firmware site in WLED00 file folder which header file or which function are used to receive data from android app

please share details which function or variable use to receive data

blazoncek, 2022-10-10:
Android (or iOS) app is just a web browser wrapper with discovery feature. Use HTTP (either JSON API or HTTP API) to communicate with WLED device. Check [KB](https://kno.wled.ge).
For further help or support questions please use WLED [forum](https://wled.discourse.group) or Discord.
Closing.

closed 2022-10-10 by blazoncek

govind183, 2022-10-10:
thank you for your reply ,
can you share some more details about which header file or function is use for this http or json api.
for better understanding for me.