wled/WLED #2824: Support for physical DMX In

martinkoenig, 2022-10-10:
**Is your feature request related to a problem? Please describe.**
No, it´s not a problem but a "nice to have". Wait, maybe there is a problem. I´m getting unexpected flashes in color palettes over ArtNet which I couldn´t figure out. It seems to be an DmxControl related problem as stated by blazoncek in issue #2811 and I still haven´t figured out why this happens

**Describe the solution you'd like**
It would be awesome if WLED would support DMX In via an physical connection using MAX485 modules

**Describe alternatives you've considered**
Well, WLED already supports ArtNet and E1.31 and it´s awesome! But a wired connection would be more reliable for WLED as lighting fixtures. My plan is to build PAR lights using 241 Bit NeoPixel Rings (the whole disk, not just a single ring)

**Additional context**
Is this possible or worth developing?

Thank you! Love you guys!

Aircoookie, 2022-10-12:
This seems like a worthwhile addition I would likely accept! (though I lack physical DMX hardware to implement myself)
(another alternative to increase reliability is of course Ethernet)

johanneswilkens, 2023-01-01:
This would be a great addition! This way we could use all different software or even physical dmx controllers to control pixels. I personally use the sweetlight software which seems to have some trouble with artnet, so a hard wired connection would be a great addition, and of course would increase reliability as well.
I see there are already 3 issues open on this topic (this one, #2594, and #2097) is there already a plan for implementing this? I would love to help realizing this feature, sadly I have limited experience in writing software, but where possible I would like to do something.

Jredbart, 2023-02-04:
Hi There,
I would also like to see the implementation of Physical DMX inputs. And am happy to assist in development.
Additionally, I noticed that there was another ticket raised (#2594) where the creator also mentioned that were happy to help with development.

Please can we get this feature pushed into development?

Kind regards.

netmindz, 2023-06-28:
This feature exists in the MoonModules fork of WLED. It uses a library that needs a newer version of the esp32-arduino platform so it's only in the V4 builds and until the platform version is updated here, I can't submit a PR.

If you want to see my changes to see how perhaps support could be added here using the current library, see https://github.com/MoonModules/WLED/pull/28

roboraptor, 2023-07-30:
How does this look? 
I only have 4 PAR lights and dedicated wifi AP for them. One of them working as a Master and DDP to others feels slow and buggy someimes. But still most of the time, does the job.
Running cables to every light is too much as it is my portable ligh instalation. 

But being able to additionaly daisychain them together with some wired platform would be awesome. <3

McKaiver, 2023-09-01:
I'm also looking for a wired DMX input... @Aircoookie i'm Based near Hamm, i could create some Hardware and give it to you for Experiments...

As important function, it should be possible to change Effekts with one Chanel.

netmindz, 2023-09-02:
You should just use the MoonModules fork of WLED if you want this feature @McKaiver

jeryci, 2024-01-09:
hi all any news on this ? we use dmx on theatre projects and as a geek i'd be so nexcited to create my own dmx fixtures (lights) with wled and include them in the light shows.. im thinking of setting up some led strips with wled and control them through dmx for setting colours , dimming, and some preset modes... maybe i can help with the testing or anything else ?

jredbart, 2024-01-26:
@jeryci Did you try the MoonModules fork as suggested by @netmindz