wled/WLED #2827: Need wifi setting to not use 11n wifi

magiva, 2022-10-11:
**Is your feature request related to a problem? Please describe.**
There are many wifi routers with 11n connectivity issues, mainly asus.  My WLED can take 10+ minutes before getting on to the wifi

**Describe the solution you'd like**
With Tasmota we have an option to choose b/g or b/g/n etc.  Id like this option in WLED

**Describe alternatives you've considered**
none

**Additional context**
none needed

Thank you for your ideas for making WLED better!

blazoncek, 2022-11-01:
To save us some time, please provide a direct link to sample code.

magiva, 2022-11-01:
I dont have sample code, i was simply referring to tasmota (which is open source) that has the ability to restrict the wifi options which inturn allows my devices to connect to my asus mesh network that doesnt like 11n.
For sample code, visit https://tasmota.github.io/ 
I refer to the WIFI command used in a console:
Wifi	0 = disable Wi-Fi
1 = enable Wi-Fi (default)
ESP8266 only:
2 = Wi-Fi mode 802.11b
3 = Wi-Fi mode 802.11b/g
4 = Wi-Fi mode 802.11b/g/n

I use WIFI 3 to avoid n

blazoncek, 2022-11-01:
I said "to save us time".
Unfortunately I do not have the luxury to browse and search in some other code.

magiva, 2022-11-01:
much that i enjoy the flipant responses of interneters, i dont really care if you do anything at all, especially as you seem unable to conduct yourself politely within a nice chat.

closed 2022-11-01 by magiva

blazoncek, 2022-11-01:
Apparently you have trouble comprehending nice requests. I said "please" and asked for help providing a link to code that does what you require as to speed up development. And as someone with 1200 commits to WLED repository I am not a random interneter.

Being "unable to ..." is what you said for yourself.

magiva, 2022-11-01:
hahah, sure.  Either way, your comment was badly worded and just plain
impolite.  I was just helping out with a suggestion.  You were unable to be
nice, kind or polite.
I just decided not to just roll my eyes and therefore commented back.

blazoncek, 2022-11-01:
Not everyone is an English speaking person.
And please point me to impolite words.