wled/WLED #2828: Add SH1107 OLED driver support to usermod_v2_four_line_display_ALT

Drizzt321, 2022-10-11:
**Is your feature request related to a problem? Please describe.**

I have [this adafruit 128x64](https://www.adafruit.com/product/4650) OLED with buttons, and it uses SH1107 driver, not SSD1306 which is what is currently supported in the [v2 Four Line Display & ALT](https://github.com/atuline/WLED/tree/master/usermods/usermod_v2_four_line_display_ALT] user mod. Trying to dig into the code, looks like if I can find the right constructor I can simply add a new type, and create a U8X8 and it Should Work(tm).

Looks like `U8X8_SH1107_64X128_SW_I2C` and `U8X8_SH1107_64X128_HW_I2C` are what I need. Unfortun

**Describe the solution you'd like**

Add support for the Adafruit SH1107 4-line OLED with buttons https://www.adafruit.com/product/4650

blazoncek, 2022-10-14:
Unfortunately I do not have a specimen to test and update code.

Drizzt321, 2022-10-14:
@blazoncek I tried editing the `usermod_v2_four_line_display_ALT.h`

Line 111, adding 
`  SH1107_64      // U8G2_SH1107_64X128_2_HW_I2C`

In the switch to create the `u8x8`, added this to the end before the `default:`
```
        case SH1107_64:
        Serial.println("Setting SH1107");
          if (!isHW) u8x8 = (U8X8 *) new U8X8_SH1107_64X128_SW_I2C(ioPin[0], ioPin[1]); // SCL, SDA, reset
          else       u8x8 = (U8X8 *) new U8X8_SH1107_64X128_HW_I2C(U8X8_PIN_NONE, ioPin[0], ioPin[1]); // Pins are Reset, SCL, SDA
          break;
```

I end up getting that output statement on boot (viewing via USB Serial).

Based on your knowledge of the library, should that work? 

I'm using [this Adafruit ESP32](https://www.adafruit.com/product/3405), in the usermod setup for this, 1st is PIN 23 (clk) and second is 22 (data). Pretty sure those would be the correct pins, right? If it'd help, I can buy and send you one, they aren't all that expensive for me.

blazoncek, 2022-10-15:
Just duplicate lines where SH1106 is replacing with appropriate constructor for SH1107.

Please join Discord and continue discussion there.