wled/WLED #2832: ESP32 build

Doyle4, 2022-10-14:
Im wanting to flash 0.14 to try with a mic and matri panel.
Some reason when I compile with ESP32_dev, doesn't show in wifi list, the led on the board blinks.
Which do I use please? Iv flashed the esp32 bootloader and nothing still.

Thanks

redPrint7, 2022-10-15:
This means ("blinking led") usually that the ESP32 is constantly resetting. Switch on the serial monitor an you'll see the RST(6,3). So far I could not get version 0.14 to run either on ESP32 or ESP8266. Depending on the config it resets constantly or when calling the web interface (Firefox on MacOSx). Compiled with Arduino IDE and platformio as well. Since 13.3. its also hard to get it run stable. Imho - to much feature creep and to much "spagetti code" in the code base. Its an IF DEF and IFNDEF hell to configure and some of the used libraries are also not well designed to keep it polite.
Just my 2 cents

Doyle4, 2022-10-15:
Thanks for replying.
I understand 0.14 is beta, i’ll make a bug finding etc post. 
I’ve managed flash a d1_mini fine, few bugs on some effects also which again understandable as its a beta.
Least I know its nothing i’m doing wrong I don’t think with the esp32z

blazoncek, 2022-10-15:
@redPrint7 you are welcome to sanitize the code. 😉 

@Doyle4 if you are unfamiliar with compiling use binaries provided by @srg74 from his Wemos shield repository on Github.

Other than that, please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
I've posted several PIO environments on Discord which should give you a working copy of 0_14 branch including audioreactive stuff.
Closing.

closed 2022-10-15 by blazoncek

Doyle4, 2022-10-16:
Thanks,

Where abouts am I looking on Discord for the files? had a quick look on each section and couldn't see anything.

Thanks again.

sansillusion, 2022-10-18:
I too have been trying to download 0.14 but can't find it at @srg74
<https://github.com/srg74> repo and don't have a discord account. I will
try to compile it using PIO but have been having issues understanding how
to get 0.14 to compile. Also my laptop is dying so I use it the least
possible and mostly use my phone to upload pre-compiled binaries.

blazoncek, 2022-10-18:
@sansillusion 
https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware
https://github.com/srg74/WLED-wemos-shield/tree/master/resources/experimental