wled/WLED #2836: WLED should have dedicated DDP mode

simmonmt, 2022-10-18:
**Is your feature request related to a problem? Please describe.**

WLED accepts DDP commands, acts on them, then ~immediately goes back to the programmed effect. This is fine if the DDP commands are milliseconds apart, but causes problems when they're separated by seconds. For example if you're using something like [pixel mapper](https://github.com/aaknitt/pixel_mapper) to locate pixels in 3D space for use in xLights. Pixel mapper turns on a pixel, waits a fraction of a second, captures an image, does CV, saves the coordinates, and repeats for the next pixel. If it can't find the pixel it leaves the pixel on while the user clicks the pixel location. These actions are easiest to implement when DDP commands are persistent -- when you can say "turn on pixel 3" in a single packet and have that persist for seconds at a time.

**Describe the solution you'd like**

I'd like a button in Sync settings that lets me say WLED will be controlled by DDP **and nothing else**. WLED should never use its builtin effects when that button is clicked.

**Describe alternatives you've considered**

A stream of DDP commands could be sent, rather than a single command. This would require either a complicated event loop or a background thread, though, which would be messy -- especially in Python.

**Additional context**

None

Thank you for your ideas for making WLED better!

blazoncek, 2022-10-18:
Unfortunately this is not going to happen but you can send JSON API command `{"live":true}` to switch to live receiving mode indefinitely.
You can create a preset with that as well.

closed 2022-10-18 by blazoncek

simmonmt, 2022-10-18:
Perfect. That will do nicely.