wled/WLED #2844: Usermod BH1750 v2 checkmark disappearing

srg74, 2022-10-23:
### What happened?

Usermod BH1750 v2 checkmark "enabled" after saving new values disappearing, but usermod is works.

### To Reproduce Bug

Change values for usermod and save. After visiting Usermods page checkmark is indicating that usermod is not enabled.

### Expected Behavior

Check mark must retain previous state.

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.0-b0 (build 2210220), WLED 0.14.0-bl0 (build 2210221)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-10-23:
@albarlow can you please verify.

albarlow, 2022-10-24:
@blazoncek I've managed to reproduce the bug, I've opened a pull request with the fix.