wled/WLED #2845: Two or more WiFi settings

Juris3D, 2022-10-23:
Would be very handy to have at least two WiFi settings for WLED.
Very inconvenient when work-in-progress device needs to be set up in several physical places (for example, home and my workplace)

chatziko, 2023-02-27:
The most flexible solution would be to accept a list of wifi configurations, each of which could be by SSID or BSSID (#2151), with a preference between them (possibly implicit preference by their order). WLED should connect to the first available one.

Another useful and relatively common use case: Many people have multiple access points, and WLED often fails to connect to the closest one. I'd like to setup:
- one configuration to a specific AP (BSSID), to be used by preference
- a fallback configuration with only SSID, to use any available AP if the main one is down.

Something like what esphome allows: https://esphome.io/components/wifi.html#connecting-to-multiple-networks

BigBill2000, 2023-06-02:
Plus one on this - I have WLED on my motorcycle so my need is a little unique. When I am at home I want to connect to my home network for updates and config etc. But when I ride with my friend who also has WLED lights, I want them to connect together and sync over WIFI. It would be nice to save both SSID's so it will be easy to connect together. They both fallback to their own AP when not at home, so I will have to change one to connect to the other. Going to impress some folks with the motorcycles side by side showing the same light show will be awesome! Thanks

JPZV, 2024-01-18:
Hi! I just added this feature in my latest PR: #3703

Please, let me know if this works to you, and don't hesitate to report me any issue that you find ;)

Juris3D, 2024-01-19:
Wow that's is great, if so! (I mean added double Wi-Fi) I will report back as soon as I try (a bit busy now, again, with Unreal Engine projects)

closed 2024-03-24 by blazoncek

Joeboyc2, 2024-12-07:
Completely missed the plus button under the wifi ip address boxes lol, that's for the work on this

Aaroneisele55, 2025-01-24:
Is there the possibility to set which WiFi the device should prefer?
I have a backup WiFi and a main WiFi and want it to only use the backup one when the main is down....

blazoncek, 2025-01-24:
The order of entries determines preference.
There is a second level of preference though: signal strength.
If the signal strength of lower entry is several magnitudes stronger, it will be preferred.