wled/WLED #2847: Restoring cfg.json does nothing

blazoncek, 2022-10-24:
### What happened?

Restoring configuration (`cfg.json`) using Backup & restore feature of Security Settings page does not restore configuration.

### To Reproduce Bug

Backup configuration on one device and try to restore it on another device.
After the upload, "File uploaded." message appears on screen (instead of "Configuration restore successful. Rebooting."). Clicking Save & Reboot reverts to previous configuration.

### Expected Behavior

Configuration restored.

### Install Method

Self-Compiled

### What version of WLED?

0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Relevant log/trace output

```shell
The problem lies in line 55 of `wled_server.cpp` (`if (filename == "/cfg.json")`) where filename is incorrectly compared.
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2022-10-24:
Just also had a look at the code. I think the problem was introduced with this commit:
https://github.com/Aircoookie/WLED/pull/2831 "fix: filename uploads"

Instead of comparing against "/cfg.json" we should also check for "cfg.json". Also it might be better to use 
```c++
if (filename.equalsIgnoreCase("/cfg.json") || filename.equalsIgnoreCase("cfg.json")) {
```

also putting the `doReboot =true` after `request->send(...)` could help, to ensure that the reboot does not happen before thereply was sent.

blazoncek, 2022-10-24:
You changed handling of original request (by prepending /). Handle that in subsequent callbacks.
filename has to be in lowercase when saving. adjust for that as well.

softhack007, 2022-10-24:
Im not sure if LittleFS is case-sensitive ... so it might be sufficient to use `filename.equals()`
The WLED web UI only allows for two specific filenames when uploading - cfg.json and presets.json.

Everything else is "extra sugar". To be 100% bullet-proof, you would also need to:
* convert files to lowecase
* strip leading and trailing blanks
* make sure filesname size is below the allowed limit (32? chars)
* reject file upload to forbidden locations
* etc etc
* and maybe present file errors to the user -> `strerror(errno)`.

blazoncek, 2022-10-24:
@softhack007 yes to all points. /upload was originally created just for the purpose of config an presets upload and then used for ir.json, skin.css, etc.
All of the logic was done in the browser in JS as to reduce ESP strain. Prepending / broke things.

closed 2022-10-24 by blazoncek