wled/WLED #2848: AZ-Delivery Mini D1 ESP32

pwcutajar, 2022-10-24:
Hi

I am trying to install WLED to a MINI D1 ESP32. I would like it to be sound reactive
 Whatever way I flash it I get the following.

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x08030403
invalid header: 0x08030403
invalid header: 0x08030403
invalid header: 0x08030403
invalid header: 0x08030403
invalid header: 0x08030403
invalid header: 0x08030403
invalid header: 0x08030403
ets Jul 29 2019 12:21:46

Any Help will be great

Thanks

Paul

blazoncek, 2022-10-24:
For SR use their GH repo or Discord.
You can upload regular WLED using online WLED installer and then upgrade to SR using OTA update.
Closing.

pwcutajar, 2022-10-24:
Online WLED installer hangs so no fix

softhack007, 2022-10-24:
Please join our discord to get help & support from the community.

[image] Join Discord to discuss [AirCookie's WLED](https://discord.gg/KuqP7NE).

[image] Join Discord to discuss beta testing of our [sound reactive WLED](https://discord.gg/4CQRmfR) of WLED.

blazoncek, 2022-10-24:
Github issues are not meant for resolving HW issues.
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.

pwcutajar, 2022-10-24:
Worked a treat!

Thanks!!!!!