wled/WLED #2849: AZ-Delivery Mini D1 ESP32

pwcutajar, 2022-10-24:
Hi

Please dont close to quickly

I am trying to install WLED to a MINI D1 ESP32. I would like it to be sound reactive
Whatever way I flash it I get the following. web browser install hangs.

also unable to to use sr .bin or browser install.

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x08030403
invalid header: 0x08030403
invalid header: 0x08030403
invalid header: 0x08030403
invalid header: 0x08030403
invalid header: 0x08030403
invalid header: 0x08030403
invalid header: 0x08030403
ets Jul 29 2019 12:21:46

Any Help will be great

Thanks

Paul

softhack007, 2022-10-24:
hey, please don't duplicate too quickly! You could ask to re-open the ticket, or even re-open it yourself 💡

pwcutajar, 2022-10-24:
I would if I knew how?

softhack007, 2022-10-24:
@blazoncek please help him, by closing this ticket.

closed 2022-10-24 by blazoncek