wled/WLED #2855: Set SK6812 data pin output frequency to 400kHz to fix flickering in 12v strings

PatientLightning, 2022-10-28:
**Is your feature request related to a problem? Please describe.**
A 12v string is flickering on transitions and effects because the WLED digital data output frequency is 800 kHz on an ESP8266, but operates smoothly with the original controller running at 400 kHz.

**Describe the solution you'd like**
Have an option/checkbox to set the digital output frequency to 400 kHz, or half speed, on the LED configuration page so higher power LEDs have a chance to read, process, and relay the commands before they get jumbled and start to flicker the whole string.

**Describe alternatives you've considered**
Tried hardware fixes like changing ESP boards, level shifters, data line resistors, and power line capacitors. Since the original controller that came with the lights works smoothly at 400 kHz, so the next best thing to try would be a software change to slow down WLED to drive the lights at 400 kHz data too. This could even fix many other 12v string flickering issues that are mentioned around the web.

Thanks!

blazoncek, 2022-10-28:
You may want to try https://github.com/Aircoookie/WLED/tree/400k-rgbw
Please report back.

PatientLightning, 2022-10-28:
Very cool, appreciate the quick reply and fresh branch.
I have pulled the branch, compiled it in vscode/platformio, and uploaded it to my esp8266 D1 mini, but I can't get the 400 kHz RGBW setting to "stick" in the web UI.
It appears to process the page when I hit save/submit, but there is no output on the pin (GPIO2) and the settings are cleared upon refresh.
I'm looking at the html to see if there's some validation check that it is failing maybe. Do you have any pointers for where I should be looking? I'm trying to understand how settings_leds.htm works because I suspect the issue is somewhere in there. Maybe trySubmit(e)? I'm trying to figure out how to setup a debugger or something to step through the code or at least send debug messages to serial so I can see how far it's making it along the process.
Thanks!

blazoncek, 2022-10-28:
Will have a look later. Perhaps I overlooked something.

blazoncek, 2022-10-28:
Fixed.

blazoncek, 2022-10-29:
@PatientLightning feedback please.

blazoncek, 2022-10-31:
@PatientLightning please report or I will close the issue without merging the code as I cannot reliably test.

PatientLightning, 2022-10-31:
I appreciate the new commit!
Good news:
    The settings are sticking and I'm getting output on the pin.
Bad news:
    The LEDs are not responding, but I made a chart with data and have a theory/request.
[image]
I think the frequency is good, but I think the pulse length is much too long.

Would we be able to shorten the pulse length at 400kHz? I suspect if it were somewhere between a third to a half as long it would be perfect.
Where in the project are the pulses shaped exactly? I still don't understand how they are generated or commanded.

Thanks!

PatientLightning, 2022-10-31:
Ok, I didn't mean to close it. Sorry I'm clumsy with this still.

blazoncek, 2022-10-31:
Please talk to @makuna of NeoPixelBus as he is the author of LED driver WLED uses.
You may want to open an issue there if it turns out this is a bug.

I did not look at the options of the 400kHz methods to be honest.

Makuna, 2022-11-03:
400kHz just runs the clock for i2s (DMA) at half the speed.  So it will maintain the normal ratio of 25% for data bit 0, 75% for data bit 1.  

Whatever LEDs you are using are not following the SK6812 spec (which is 800kHz data speed).  Are you sure this is what you have?
Further, they don't even follow the normal ratio for the pulse form.  From what you posted, the data bit 1 ratio is 25%, then what is the zero ratio?  12.5%?  The esp8266 implementation only has five distinct pulse forms it can support, 0%, 25%, 50%, 75%, and 100%; of which only 3 are usable to create an actual pulse.

Are you sure the flickering is due to pulse rather than the reset time?  The SK6812 is specified to have a short one, if you use the WS2812x method instead it will use a longer reset time but have the same pulse form as the SK6812 method.  If you generally see the right colors but see flickers, this reset time is the usual culprit.

blazoncek, 2022-11-03:
@Makuna thank you for taking the time.

blazoncek, 2022-11-18:
As this is LED driver issue or unsupported LED chip (non-standard implementation) I am closing this issue.
I will keep the `400k-rgbw` branch alive for a while. Maybe even merge it, if it proves useful.

closed 2022-11-18 by blazoncek