wled/WLED #2857: Audio_Reactive Usermod

DSRocker, 2022-11-01:
### What happened?

compiling does not work after customizing the platformio.ini:

build_flags =
  -D USERMOD_AUDIOREACTIVE
lib_deps =
    arduinoFFT 

This is the error:
wled00/../usermods/audioreactive/audio_reactive.h:256:51: error: no matching function for call to 'arduinoFFT::MajorPeak(float*, float*)'

### To Reproduce Bug

customizing like written here: https://github.com/Aircoookie/WLED/tree/main/usermods/audioreactive and compile 

### Expected Behavior

no errors

### Install Method

Self-Compiled

### What version of WLED?

the last version

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-11-01:
Please use: https://github.com/blazoncek/arduinoFFT.git for `lib_deps`. It is a float only implementation.

softhack007, 2022-11-01:
I think this is only a documentation issue.

You need either 
```ini
build_flags = 
    -D USERMOD_AUDIOREACTIVE
lib_deps = 
    https://github.com/blazoncek/arduinoFFT.git#master
```
or 
```ini
build_flags = 
    -D USERMOD_AUDIOREACTIVE -D UM_AUDIOREACTIVE_USE_NEW_FFT
lib_deps = 
    https://github.com/kosme/arduinoFFT#develop @ 1.9.2
```

DSRocker, 2022-11-01:
Thanks a lot for your very fast answers! I'll try it!

I downloaded the arduinoFFT lib and put the files into the lib dirictory of WLED.. It seems to work, too...
But I'll try your suggestions.

softhack007, 2022-11-01:
don't use the "plain" arduinoFFT lib release. It's it uses double and is extremely slow. The two options avove are both faster, and no manual download needed. Please also delete the  arduinoFFT lib you downloaded, otherwise platformIO might get confused.

DSRocker, 2022-11-01:
OK! Thanks guys!!!

DSRocker, 2022-11-01:
Great. Seems to work; now it*s time to connect all the wires and test it.
But: in Usermod there is now some configurations:
Global I2C GPIOs (HW) SDA / SCL
Global SPI GPIOs (HW) MOSI / MISO / SCLK
Do I need to configure them?!

Addition to Audio_Reactive I use also Temperature and MultiRelay Usermod.

blazoncek, 2022-11-01:
You don't need to configure.
Please close if your OP is solved.

closed 2022-11-01 by DSRocker