wled/WLED #2862: Timed Light Default Duration Override is Persistent

danmalone326, 2022-11-03:
### What happened?

I have a regular set of timed lights throughout the day, so I have set the timed light default duration to 60 minutes under LED Preferences. 

[image]

This works fine when I use the API (&ND) or the web interface to turn on the timer.  The popup on the web interface says:
   
[image]
 
Occasionally, I need the timed light to last for 1 minute, so I use the API to start a 1 minute timed light (&NL=1). This also works fine.

However, the next time I want to start a timed light with the default duration, using the API (&ND) or the web interface, it starts a 1 minute timed light instead of the default 60 minutes. The popup on the web interface says:

[image]

After rebooting, default reverts back to 60 minutes.

### To Reproduce Bug

set timed light default duration to 60. (maybe make this 5 to see it quicker)

http://172.16.5.148/win&ND
<vs>
<ac>127</ac>
<cl>255</cl>
<cl>160</cl>
<cl>0</cl>
<cs>0</cs>
<cs>0</cs>
<cs>0</cs>
<ns>0</ns>
<nr>1</nr>
<nl>1</nl>
<nf>1</nf>
<nd>60</nd>
<nt>0</nt>
<fx>0</fx>
<sx>128</sx>
<ix>128</ix>
<fp>0</fp>
<wv>-1</wv>
<ws>0</ws>
<ps>0</ps>
<cy>0</cy>
<ds>WLED-Lamp</ds>
<ss>0</ss>
</vs>

Time to see how long it takes to complete.

http://172.16.5.148/win&NL=1
<vs>
<ac>125</ac>
<cl>255</cl>
<cl>160</cl>
<cl>0</cl>
<cs>0</cs>
<cs>0</cs>
<cs>0</cs>
<ns>0</ns>
<nr>1</nr>
<nl>1</nl>
<nf>1</nf>
<nd>1</nd>
<nt>0</nt>
<fx>0</fx>
<sx>128</sx>
<ix>128</ix>
<fp>0</fp>
<wv>-1</wv>
<ws>0</ws>
<ps>0</ps>
<cy>0</cy>
<ds>WLED-Lamp</ds>
<ss>0</ss>
</vs>

Time to see how long it takes to complete.

http://172.16.5.148/win&ND
<vs>
<ac>122</ac>
<cl>255</cl>
<cl>160</cl>
<cl>0</cl>
<cs>0</cs>
<cs>0</cs>
<cs>0</cs>
<ns>0</ns>
<nr>1</nr>
<nl>1</nl>
<nf>1</nf>
<nd>1</nd>
<nt>0</nt>
<fx>0</fx>
<sx>128</sx>
<ix>128</ix>
<fp>0</fp>
<wv>-1</wv>
<ws>0</ws>
<ps>0</ps>
<cy>0</cy>
<ds>WLED-Lamp</ds>
<ss>0</ss>
</vs>

Time to see how long it takes to complete.

### Expected Behavior

Documentation says &ND "Toggles nightlight on but uses default duration".
&ND or clicking the timer button on the web interface should use the timed light default duration.

### Install Method

Binary from WLED.me

### What version of WLED?

Server message: WLED 0.13.1 (build 2203150)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-11-04:
As a workaround have you considered changing your `&ND` into `&NL=60`?
Otherwise this is a confirmed discrepancy between documentation and implementation.

If you want to test the fix yourself add line 955 in `set.cpp` like this:
[image]
```c++
nightlightDelayMins = nightlightDelayMinsDefault;
```

blazoncek, 2022-11-04:
Commit 71d84fecba1daccf43c9319ee18b3f149156eb6a should fix this.

closed 2022-11-04 by blazoncek

danmalone326, 2022-11-04:
Thank you for the quick response! 
Similar to the workaround you suggested, I'm using the Home Assistant integration and Node Red, but this requires me to use MQTT to access the API and the &NL=60 .
I'm looking for were to submit a feature request for the integration to allow a parameter on the Nightlight Switch.

blazoncek, 2022-11-04:
> I'm looking for were to submit a feature request for the integration to allow a parameter on the Nightlight Switch.

That would be @frenck most likely.