wled/WLED #2864: Slower effects for garlands

mitrokun, 2022-11-04:
Hi, one more request)
When using string garlands (10 LEDs per meter), many animations happen too fast even at the minimum speed level. Is it possible to add a checkbox (or a text field to indicate the LED density per meter) in the settings, which will add a deceleration factor to the standard speed?
For the checkbox option, slowing down x10 will be enough.
Or, if use a text field, wled  can recalculate the speed relative to standard strips with a density of 120 LEDs per meter.

blazoncek, 2022-11-04:
Thanks for the suggestion.
If you can write a generic *algorithm* how to achieve that I will be happy to implement it. 😄 

Unfortunately effect functions have no knowledge of pixel density and hence no way to adjust speed accordingly.
You can, however, write down a list of all effects that look *too fast* and I will try to adjust their minimum speed if possible.

mitrokun, 2022-11-04:
If you manage to slow down the Colorwaves, it will be a big deal. In my opinion, this is the most necessary effect for creating ascending or descending streams of light.

blazoncek, 2022-11-04:
Looking at Colorwaves on 10 pixel strip looks quite nice at low speeds.
But I am not sure if that is what you mean. Can you post a video of what you want to achieve?
I encourage you to join Discord or WLED [forum](https://wled.discourse.group) and continue discussion there.

mitrokun, 2022-11-04:
https://www.youtube.com/shorts/mPUnUT0qdyw
Here are my two-meter ribbons. A little fast)

you can use these settings for the test - speed 0, intensity ~122, colors 1&2, black and any other

508loopdetect, 2023-11-30:
i also have the problem on a almost all animations that they look still  way too fast on 1%. 

if there is a checkbox or just a script option to use it in a preset config, this would make me more than happy.

elpollodiablo, 2024-02-17:
I'm currently having the "that's way too fast" issue too, but with a single bulb. I want that "very slow change over time" effect that professional mood lighting has, i.e. the change should be basically not noticeable. Not quite like watching grass grow, but almost. I would be extremely happy to see such a mode for at least the colorchange/fading animations in wled.

blazoncek, 2024-02-17:
There are over 180 effects, you need to be specific.

508loopdetect, 2024-02-18:
at first i have to say some things changed with the new 14 version. some effects seem to be slower like in older versions, is this possible?

 it seems that the effect slider and speed slider changed function  
used colortwinkle and sakura. speed 0 is almost white, above speed 4 it turns more into red and does what you expect. 
maybe just my installation, but.... was not before. i use newest stable Hoshi

i only use my presets at speed 0, because higher speeds are not looking smooth in a big installation.
I use 3 rows, each with 3x30 leds  60pcs/m strip type. 

effect candle, multi candle is slower than others changing the leds. 
its ok with speed 0, but could be slower. 
i also miss a animation or function to use the single 50Watt  2812led from a german shop. 

**the faster and not nice or realistic looking effects are:**
Bouncing Balls at 0. Much too fast, annoying at 115 
BPM at 0 could be a bit slower, its a fast animation though
colorwaves at 0 is 10 times to fast, avove 50 a machine gun
Pacifica could be a bit slower 

**Speed slow enough to use as is:**
Aurora at speed 0 is okay, i remember that one much more faster in versions before, because its one of my favs.
colorloop is ok at 0, but i will benefit from being much slower at big installations with smooth changes
Dancing shadows is ok at 0

**Perfect one**
Glitter with Default Color.  this has the best range regarding speed from 0 to 255.
with 0 it has almost no movement and  255 will serve stripes with high led density where you need it.

Minimemat, 2024-02-19:
I agree with the slowing of almost every effect. I tried almost all of them and barely came out with any good ones (although all of them are good) because they wont go slower. Would be willing to pitch in if I can

blazoncek, 2024-02-19:
Just to satisfy my curiosity and to get the sense of what you consider too fast, can anybody share a video of (at least) one of such effects with the slowest speed possible (and a slowed down video of desired speed for comparison)?

Minimemat, 2024-02-21:
The 2 different lines connect at the center post. This is only 1 but the same can be said for almost any pattern that is linear or travelling. The LEDs are 9" apart which is much larger than the normal setup. I know there's a way to modify the pattern directly but there's a ton. I can definitely do some of the grunt work and list out all of the patterns if needed too. 

https://github.com/Aircoookie/WLED/assets/12836289/106eba4c-0641-4c44-aaed-a8138d359a77

508loopdetect, 2024-02-22:
what was the speed setup there? i guess it is lower 10. 
i will also link one or more videos next days, when my gimbal is here.

Minimemat, 2024-02-22:
> what was the speed setup there? i guess it is lower 10. i will also link one or more videos next days, when my gimbal is here.

The left is wled chase 2 on the slowest setting 0/1 the right is the speed of most led controllers on a similar effect on the lowest speed but this is on a lot of effects like you said before

mitrokun, 2024-03-25:
> slowed down video of desired speed

original at minimum speed

https://github.com/Aircoookie/WLED/assets/108380896/ae83b5ec-03fd-4570-8c5a-0f9b6db20a6b

The desired speed. Three times slower

https://github.com/Aircoookie/WLED/assets/108380896/d4cc99ed-b416-4369-aefb-1314e8265d4d

Minimemat, 2024-03-26:
[FX.txt](https://github.com/Aircoookie/WLED/files/14762063/FX.txt)

I updated the FX file and modified almost all of them so the slider has a larger range and slower. V0.14.1

blazoncek, 2024-03-26:
Please make a PR and base it off of 0_15 branch.