wled/WLED #2866: iOS App not available in Austria/Europe

bonderaustria, 2022-11-04:
### What happened?

Hello!

I wanted to install the WLED app from the appstore, but i couldnt finde it. After 10 minutes i found a direct link to the app, if i clicked on the link it said: The app is not available in your country/region

### To Reproduce Bug

Publish the app again for austria:)

### Expected Behavior

The app should be available again:)

### Install Method

Binary from WLED.me

### What version of WLED?

the latest 

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Aircoookie, 2022-11-04:
The iOS app is temporarily unavailable globally due to an administrative situation between me and Apple. I hope it will be available again within the next few hours. Sorry for the inconvenience :)

Aircoookie, 2022-11-07:
Will close once the app is available again :)

bonderaustria, 2022-11-07:
Now the app is back online:)

closed 2022-11-07 by bonderaustria