wled/WLED #2867: Update HA Entities when adding or removing presets or playlists

Wombosvideo, 2022-11-04:
### What happened?

Whenever I add a new preset in the WLED UI the Home Assistant entity does not get updated which makes it impossible to select the newly saved preset. The same goes for playlists.

### To Reproduce Bug

1. Integrate WLED in Home Assistant
2. Create a preset or playlist in the WLED UI

### Expected Behavior

The entity is updated in Home Assistant so that the newly saved preset or playlist can be selected.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.3 (build 2208222)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-11-04:
This is HA integration issue, not WLED.
Please consult HA team.

dosipod, 2022-11-07:
It is an HA issue which is not limited to wled integration , to see the preset or  playlist in most cases just reload wled integration

Wombosvideo, 2022-11-08:
An issue has been opened at the right place then:
https://github.com/home-assistant/core/issues/81770