wled/WLED #2869: TM1814 Bug

mrrobertkent, 2022-11-06:
### What happened?

Whenever changing brightness, color, animation, turning on , or turning off the lights. The light strip blinks red before the change takes affect. 

### To Reproduce Bug

This problem seem to persist accross multiple different pieces of hardware. I've test on a Dig Quad and Uno as well as several different TM1814 light strips.

### Expected Behavior

The leds are not supposed to blink red when there is a change.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.1 (build 2203150)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-11-06:
Set black color for primary slot (1) and test again. You can also experiment with transition time (set it to 0 or 10s and observe).

If there is no red flash then, it means you are seeing color transition from one effect to another.

Some effects do not use color slots but palette instead. This is where the problem comes from.

blazoncek, 2022-11-18:
If there are no updates I will assume issue is solved.

closed 2023-03-27 by blazoncek