wled/WLED #2871: Matter Update possible?

derpate1989, 2022-11-06:
Hi, is a Matter Update possible? :) That would be a huge gain for Wled and various smart home systems.

Aircoookie, 2022-11-06:
Matter support would definitely be great as it might just be the missing link to make WLED devices work locally with all common smart speakers.
Right now I see three main issues, which I hope can be solved:
- There should be an Arduino library for it ([this](https://github.com/jakubdybczak/esp32-arduino-matter) seems like a starting point already)
- It requires encryption, and unless the matter specification allows self-signed certs (which I doubt), it will be a pain to have a PKI infrastructure. Also, due to memory constraints, this will rule out matter ever working on ESP8266.
- I am not sure whether Bluetooth provisioning is optional or a core requirement of the Matter specification. Bluetooth on ESP32 is supported, but uses an extreme amount of memory and storage, which is why we steered clear of using it within WLED thus far.

derpate1989, 2022-11-07:
Thank you for this detailed explanation! I have a lot of hope in Matter and look forward to this standard. Maybe someday there will be a way for Wled and Matter :)

DoomHammer, 2022-11-10:
> Matter support would definitely be great as it might just be the missing link to make WLED devices work locally with all common smart speakers. Right now I see three main issues, which I hope can be solved:
> 
> * There should be an Arduino library for it ([this](https://github.com/jakubdybczak/esp32-arduino-matter) seems like a starting point already)
> * It requires encryption, and unless the matter specification allows self-signed certs (which I doubt), it will be a pain to have a PKI infrastructure. Also, due to memory constraints, this will rule out matter ever working on ESP8266.
> * I am not sure whether Bluetooth provisioning is optional or a core requirement of the Matter specification. Bluetooth on ESP32 is supported, but uses an extreme amount of memory and storage, which is why we steered clear of using it within WLED thus far.
> 
> Still I am enthusiastic for the Matter standard, if only for realizing that a unified smart home standard is necessary and that mandatory device vendor clouds are a terrible idea.

If it's not possible to do entirely on a device, maybe a bridge could help here?

Aircoookie, 2022-11-10:
That's a nice idea, it should definitely be feasible to make some software to act as a bridge between the JSON API and Matter protocol(s), and it would even support ESP8266 units. Native support would be ideal though.

jasta, 2022-11-25:
> - I am not sure whether Bluetooth provisioning is optional or a core requirement of the Matter specification. Bluetooth on ESP32 is supported, but uses an extreme amount of memory and storage, which is why we steered clear of using it within WLED thus far.
> 

I don't believe it is.  The spec says that there are 3 methods, BLE, WiFi SoftAp (which no one is using today unfortunately), but also existing DNS-SD if you're already on the local network.  This last case seems perfect for WLED which is already so easy to get setup to connect to WiFi.

It seems that it just needs to register some mDNS records like _matter._udp/_tcp maybe?  I'm still trying to get full access to the spec to read the details myself and start playing around with this a bit.

Chase295, 2023-07-04:
Is there any new stuff or soultion a "bridge" or any think else becous i would like to add it in the nanoleaf app for the new Nanoleaf 4D

blazoncek, 2023-07-05:
Unless someone with knowledge and programming skills joins in there will be no progress.

heho0011, 2023-07-10:
Hi,
maybe Matter can be implemented the same way as in tasiota for esp32?

Stoffe, 2023-08-09:
Seems like Espressif has a new esp32 with support?
https://www.espressif.com/en/news/ESP32-C6_Available

And using Yacubane esp32 lib, you could maybe implement matter in esp32 or take ideas from...
https://github.com/Yacubane/esp32-arduino-matter
Here is a video of the install, https://www.youtube.com/watch?v=HpACspFygUI

might be something to look into...
I would also upvote this feature for sure.

marcoscv-work, 2023-09-15:
I have seen that tasmota already has support for Matter. I've been testing and works fine in homekit. it's interesting because they provide ESP8266 support by ESP32, Looks promising and a really good idea.
https://tasmota.github.io/docs/Matter/

Right now all solutions to integrate WLED in homekit are pretty painful, implementing matter in some way would be awesome.

jonahnm, 2023-09-29:
I will be attempting to implement this soon.

khimaros, 2023-09-30:
@Aircoookie thank you for the detailed breakdown, a few questions inline:

>     * There should be an Arduino library for it ([this](https://github.com/jakubdybczak/esp32-arduino-matter) seems like a starting point already)

what sort of signal would you look for in this library to consider it ready for use in WLED?

i filed https://github.com/Yacubane/esp32-arduino-matter/issues/44 re: publishing to platformio registry, but i'm not sure if you consider this a blocker.

>     * It requires encryption, and unless the matter specification allows self-signed certs (which I doubt), it will be a pain to have a PKI infrastructure. Also, due to memory constraints, this will rule out matter ever working on ESP8266.

from what i've been able to find, there is no restriction placed on which CAs are used or any other verbiage limiting use of self-signed certificates in the spec.

if this were restricted, a workaround is to use LetsEncrypt with real domain name the user controls, or to trust/use a services such as https://sslip.io/ or https://backname.io/ to get valid certs with any IP address (including private blocks).

>     * I am not sure whether Bluetooth provisioning is optional or a core requirement of the Matter specification. Bluetooth on ESP32 is supported, but uses an extreme amount of memory and storage, which is why we steered clear of using it within WLED thus far.

the library you linked (https://github.com/jakubdybczak/esp32-arduino-matter) "comes with precompiled NimBLE, because default Bluedroid shipped with arduino-esp32 takes too much RAM memory", does that resolve your concern?

> Still I am enthusiastic for the Matter standard, if only for realizing that a unified smart home standard is necessary and that mandatory device vendor clouds are a terrible idea.

it would be awesome for users to be able to use their existing apps and assistants to also control their WLED devices. great for spouse approval of these devices :)

chikko80, 2024-10-27:
@khimaros @Aircoookie 

[https://github.com/espressif/arduino-esp32/pull/10467](https://github.com/espressif/arduino-esp32/pull/10467)

Maybe i am wrong here, i am relatively new to this space, but i stumbled across this merge request from [espressif](https://github.com/espressif) which was merged very recently and seems to add matter support for esp32.
Can you work with that?

softhack007, 2024-10-27:
@chikko80 thanks for the hint 👍 

Its PR for arduino-esp32 v3.1, so unfortunately it's still far away from us - we are using v2.0.9 right now for some boards.  

At the moment we cannot upgrade a newer arduino version for several reasons
* bugs in the new arduino framework - espressif needed 2 years to make v2.0.x usable and stable, then declared it unsupported once it was usable. So I expect the same for v3.1 which is not released yet.
* WLED does not compile with the new arduino framework v3.0.x - to many breaking changes introduced by espressif
* [pPlatformIO does not support the new v3.0.x arduino framework](https://github.com/platformio/platform-espressif32/issues/1225), and it's questionable if this will change.

But time will tell, and maybe in a few years from now we can use matter support provided by arduino-esp32 v3.1.x.

jonahnm, 2024-11-22:
> I will be attempting to implement this soon.

And then I forgot!

khimaros, 2024-11-23:
there is a community supported PlatformIO module with support for arduino core 3.0.x: https://github.com/pioarduino/platform-espressif32 -- see also https://github.com/espressif/arduino-esp32/discussions/10039

quickquest88, 2024-12-24:
> > I will be attempting to implement this soon.
> 
> And then I forgot!

Did you remember this time?  In either case, look, i'm still very much a programming newb, with very little knowledge, but I can learn quickly if given direction, and I am motivated to help, I really would like to see WLED over thread. I will offer any help I can.  I;m going to try and start reading up on the thread spec and libs, and get familiar, i hope.  please let me know if I can help.

M4rkoV2, 2025-07-31:
Any new progress on this?

mateuszstalmach, 2025-08-14:
Hi,

I saw that the latest Matter specification (1.4.2) no longer requires Bluetooth commissioning.

Would it be possible to add Matter support to WLED for ESP32 now, using only Wi-Fi commissioning?

I think this update could make Matter integration easier without extra hardware.

netmindz, 2025-08-15:
The Espressif Matter SDK needs the V5 but we are only on V4 at the moment, so this would be blocked by 
https://github.com/wled/WLED/pull/4838

netmindz, 2025-11-09:
See comment on how to add https://github.com/wled/WLED/issues/3078#issuecomment-3506845307

elsiehupp, 2026-04-23:
> I've not yet looked, but not sure what the advantages of Matter over Thread would be for WLED over ZigBee

Matter-over-Thread and Zigbee have similar features but are not directly compatible. For instance, [Matter-over-Thread supports binding](https://developer.espressif.com/blog/matter-device-to-device-automations/) in much the same way as [Zigbee](https://docs.silabs.com/zigbee/9.0.0/zigbee-concepts-network/binding), though Matter-over-Thread binding is not particularly well supported as of yet.

As for Matter-over-Thread versus Matter-over-WiFi, Thread is a mesh network designed for larger numbers of devices than most consumer WiFi routers, so in addition to supporting peer-to-peer binding Matter-over-Thread should also be more reliable at scale.

And as for bridging Matter to WLED... Matter-over-Thread with its peer-to-peer binding and mesh topology is just a lot more reliable. If individual devices can support Matter-over-Thread and WLED's own protocol simultaneously, though, all the better!

Note that I don't personally have any WLED devices, as I have been working with Zigbee (via Zigbee2MQTT), despite its disadvantages, largely because of Zigbee's similar benefits versus WiFi when compared to Matter-over-Thread. I found my way here to this Issue because my brother and I are in the process of setting up Matter-over-Thread lighting at our parents' house, and the options for Matter-over-Thread LED controllers, i.e. not just bulbs, are still somewhat limited compared to the options for Zigbee LED controllers. If some WLED LED controllers could directly support Matter-over-Thread (or even Zigbee), that support would open up a lot more options for our new setup.