wled/WLED #2872: Adalight missing magic word

trigu75, 2022-11-07:
### What happened?

when using AAA (prismatik) on android and trying to connect to esp32 wled (that's what I have) using adalight 

it fails to make connection with the lightstrip  

but works fine if I use an adalight device instead of wled  , doing some debug , I found that the adalight device is constantly sending Ada through serial , something wled is not doing

I checked Adalight code and found 

  } else if (!Serial.available() && (ada_sent + 5000) < millis()) {
                  Serial.print("Ada\n"); // Send "Magic Word" string to host every 5 seconds... 
                  ada_sent = millis(); 
            }

as you see in their code they send a magic word , which apparently is used by different software to recognize it is connected 

### To Reproduce Bug

you need android ambient light apk  (prismatik for android) 

and ESP32 wled connected to android through USB

### Expected Behavior

Ada\n is expected to be sent from wled over serial every 1000 miliseconds / 5000 miliseconds to become an adalight compatible hardware

### Install Method

Binary from WLED.me

### What version of WLED?

latest 0.13.3 and 0.13.3 SR

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
no log
```

### Anything else?

      // No data received.  If this persists, send an ACK packet
      // to host once every second to alert it to our presence.
      if((t - lastAckTime) > 1000) {
        Serial.print("Ada\n"); // Send ACK string to host
        lastAckTime = t; // Reset counter
      } 

this from here https://github.com/adafruit/Adalight/blob/master/Arduino/LEDstream/LEDstream.pde

and my other example from : https://raw.githubusercontent.com/sticilface/Adalight-ESP8266/master/Adalight-ESP8266.ino

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-11-07:
WLED is a receiving Adalight device (LED driver) so it only needs to inform host of its capabilities/presence once.
A quote from the source you linked is talking about sender, not receiver:
```
// A 'magic word' (along with LED count & checksum) precedes each block
// of LED data; this assists the microcontroller in syncing up with the
// host-side software and properly issuing the latch (host I/O is
// likely buffered, making usleep() unreliable for latch).  You may see
// an initial glitchy frame or two until the two come into alignment.
// The magic word can be whatever sequence you like, but each character
// should be unique, and frequent pixel values like 0 and 255 are
// avoided -- fewer false positives.  The host software will need to
// generate a compatible header: immediately following the magic word
// are three bytes: a 16-bit count of the number of LEDs (high byte
// first) followed by a simple checksum value (high byte XOR low byte
// XOR 0x55).  LED data follows, 3 bytes per LED, in order R, G, B,
// where 0 = off and 255 = max brightness.
```

Aircoookie, 2022-11-07:
Hmm, seems like it is indeed common for Adalight receivers to send the magic word every few seconds (and the app you are using seems to expect it in order to start sending data).
https://github.com/dmadison/Adalight-FastLED/blob/master/Arduino/LEDstream_FastLED/LEDstream_FastLED.ino#L222

If you reset the ESP while the adalight sender app is active, it should work as we print `Ada` once on successful boot.
Might need to change it to print every few seconds but I'm reluctant to add that as it can interfere with a) using serial for other communication and b) using the TX/GPIO1 pin for LED out (which you pretty much have to do to have more than one digital LED output on ESP8266)

trigu75, 2022-11-07:
@Aircoookie 

you are correct , Prismatik (windows and android) doesn't work , until I press a few times the reset button on ESP , it not always gets the 1st Ada msg ,but  once it's "connected" it start working
maybe adding a check mark on UI to "enable / disable" Adalight over serial (or any other option over serial) could help 
if Adalight is enabled , then Ada Magic work is sent to serial every second (if data is not received) , if Adalight is disabled , serial works like normal

Aircoookie, 2022-11-07:
Exactly, it'd need to be an option :)