wled/WLED #2873: PIR_sensor_switch Master Branch is broken

Sangoku, 2022-11-07:
### What happened?

I tried all compiler flags to no avail.   
The thing is just plainly broken by a git merge which the autor admits in the last GIT pull request. 

Either he is not mentioning somthing or i am missing somthing. 

### To Reproduce Bug

platform_override.ini file:  and it just DOES not work.. 
```
 [platformio]
default_envs = WLED_motion

[env:WLED_motion]
board = esp01_1m
upload_speed = 74880
upload_port = COM7
platform = ${common.platform_wled_default}
platform_packages = ${common.platform_packages}
board_build.ldscript = ${common.ldscript_1m128k}
lib_deps = ${esp8266.lib_deps}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp8266}
; *********************************************************************
; *** Use custom settings from file my_config.h
  ;  -DWLED_USE_MY_CONFIG
; *********************************************************************das
;
;
; *** To use the below defines/overrides, copy and paste each onto it's own line just below build_flags in the section above.
; 
; disable specific features
;  -D WLED_DISABLE_OTA
 -D WLED_DISABLE_ALEXA
 -D WLED_DISABLE_BLYNK
 -D WLED_DISABLE_HUESYNC
 -D WLED_DISABLE_INFRARED
;  -D WLED_DISABLE_WEBSOCKETS
; PIN defines - uncomment and change, if needed:
  -D LEDPIN=2
;   -D BTNPIN=0
;   -D TOUCHPIN=T0
;   -D IRPIN=4
;   -D RLYPIN=12
;   -D RLYMDE=1
; digital LED strip types - uncomment only one ! - this will disable WS281x / SK681x support
;   -D USE_APA102
  ; -D USE_WS2801
;   -D USE_LPD8806
; PIN defines for 2 wire LEDs
   ; -D CLKPIN=0
   ; -D DATAPIN=2
; to drive analog LED strips (aka 5050) hardware configuration is no longer necessary
; configure the settings in the UI as follows (hard):
;   for the Magic Home LED Controller use PWM pins 5,12,13,15
;   for the H801 controller use PINs 15,13,12,14 (W2 = 04)
;   for the BW-LT11 controller use PINs 12,4,14,5
;   
; set the name of the module - make sure there is a quote-backslash-quote before the name and a backslash-quote-quote after the name
  -D SERVERNAME="\"WLED-Bathrom\""

; enable PIR usermod
  -D USERMOD_PIRSWITCH
  -D PIR_SENSOR_PIN=0

; wifi sid:
  -D CLIENT_SSID="\"xxxx\""
  -D CLIENT_PASS="\"xxxxx\""
; enable MQTT
  ; -D WLED_ENABLE_MQTT

  -D  WLED_DEBUG
 
;   
; set the number of LEDs
;   -D DEFAULT_LED_COUNT=30
;   
; set milliampere limit when using ESP pin to power leds
;   -D ABL_MILLIAMPS_DEFAULT=850
;
; enable IR by setting remote type
;   -D IRTYPE=0 ;0 Remote disabled | 1 24-key RGB | 2 24-key with CT | 3 40-key blue | 4 40-key RGB | 5 21-key RGB | 6 6-key black | 7 9-key red | 8 JSON remote
;   
; set default color order of your led strip
;   -D DEFAULT_LED_COLOR_ORDER=COL_ORDER_GRB

```

### Expected Behavior

The esp should create a topic which describes the device. Which it does
/hassio/binary-sensor/xxxxx
{"name":"WLED-Bathrom Motion","stat_t":"wled/62c9e7/motion","pl_on":"on","pl_off":"off","uniq_id":"8caab562c9e7_motion","dev_cla":"motion","exp_aft":1800,"device":{"name":"WLED-Bathrom","ids":"wled-sensor-WLED-62c9e7","mf":"WLED","mdl":"FOSS","sw":"0.14.0-b0"}}

And the device gets registered on led/62c9e7 BUT no matter what i do it wont post to the motion topic

no matter if any combination of the checkboxes are selected or if mqtt is configured to run or not. 

### Install Method

Self-Compiled

### What version of WLED?

master lastest checkout 7.11.2022

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
When MQTT is enabled the publishing topic gets generated where the config states that the motion sensor should publish to the wled/deviceId/motion topic and that the sensor should listen to it.. 
That topic does not get generated no matter what i do.
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-11-07:
Please clean up your PIO environment.
I cannot reproduce your observation.

Sangoku, 2022-11-08:
I did. Clean Windows 10 install, VSC new all is new.
I even installed the docker dev env. and compiled it there...
Same behaviour. 

The last release, [v0.13.3](https://github.com/Aircoookie/WLED/commit/102a28aef49371a989fcfb35966cd2133d609a55)  is working fine, I see the /motion topic fine then. 
But ofc it is missing the HAS integration sadly. 

I would not open a bug task without trying everything I could think of.. Wasted 2 days on debugging 
When compiled from master the mqtt /motion topic does not get send. not once from the debugger console or in the mqtt server. 

Things i tried:
compiled with/without every
WLED_DISABLE_xxx flag
Then for every compile, tried to enable mqtt to get it to post. tht is why i hardcoded the sid and pass so i dont have to set it up everytime. 
turned on/of every flag in settings.. seems like it just won't publish to the /motion topic. 
the register topic gets posted but the referenced wled/62c9e7/motion never gets posted and the mqtt sensor just stays unavailable. 

my workflof for debugging is:
compile -> upload  to esp with esphome flasher -> reset -> see the integration being found by HASSIO -> either go to mqtt end enable it and restart/ or just add the integration.  
Expected is that it should just show up in the integration.. but it shows up only in MQTT when you enable it in the settings.

As far i get from the commits on the branch https://github.com/Aircoookie/WLED/pull/2853 they had a git merge Error, my guess is something is broken. 
Did you do git pull from master when you compiled?

blazoncek, 2022-11-08:
Please point me to a line in commit 7211e6b9293d089dab0a48c5219654db00dead79 where you think the change broke MQTT publishing (as I cannot find any line for publishing MQTT messages that was changed). The only change was to add publishing on `homeassistant/binary_sensor/<mac address>/config` topic on MQTT broker (re)connect, which informs Home Assistant of available sensor. This message is *retained* so it may always show when browsing/querying MQTT broker.

Also I do not understand where your information on "on the branch #2853 they **had git merge error**" comes from? There were no errors when merging #2853.

blazoncek, 2022-11-14:
Please comment or provide additional feedback.

blazoncek, 2022-11-25:
Not reproducible.

closed 2022-11-25 by blazoncek