wled/WLED #2874: WLED does not save wifi settings when clicking Save & Connect

spierepf, 2022-11-08:
### What happened?

I power up the board and it presents an WLED-AP access point.

I configure Wifi Settings and set Network Name and Network Password correctly.

I click on "Save & Connect" and the board connects to the network.

Then I power cycle the board and those settings are forgotten and I have to start over again.

### To Reproduce Bug

Follow the steps above.

### Expected Behavior

I expect that, once I've configured the wifi and clicking Save & Connect, that the board will connect to that wifi instead of presenting an AP.

### Install Method

Binary from WLED.me

### What version of WLED?

2208222

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

spierepf, 2022-11-08:
I tried using `esptool.py erase_flash` but with no effect.

spierepf, 2022-11-08:
I get the same result on a different board.

emiryapici, 2022-11-14:
same as mine. forgets everything on ESP8266 after reboot. also cannot save any presets

softhack007, 2022-11-15:
> build 2208222

Can you try with a more recent version of WLED? Presets and config file handling has been rewritten during the last weeks, could be that it works better for you with the latest build.

https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/WLED_wemos_shield/v0.14.0-b0

https://wled-install.github.io/

blazoncek, 2022-11-18:
There are some NodeMCU compatible ESP8266 floating around that have faulty WiFi.
Try installing WLED 0.8.x as that is the last version that works on that particular variant of ESPs.

You can also verify this by purchasing original Wemos D1 mini (currently running v4, but I have v3 and v2 running WLED successfully).

closed 2024-04-23 by github-actions[bot]