wled/WLED #2878: Won't boot when "Disable realtime gamma correction" is un-checked

shayded-exe, 2022-11-10:
### What happened?

When I un-check "Disable realtime gamma correction" under the Artnet settings, WLED will no longer boot and I have to wipe my esp8266 to get it to work again.

It doesn't matter if I'm actually sending any artnet data. Un-checking that setting completely bricks it.

### To Reproduce Bug

Uncheck "Disable realtime gamma correction" under the Artnet settings.

### Expected Behavior

WLED doesn't freeze on startup.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.13.3_ESP01

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-11-10:
Sorry, can't reproduce. Tried on ESP8266 (D1 mini), ESP32, ESP32-C3 & ESP32-S2

shayded-exe, 2022-11-11:
Is there any way I can capture log output being that I can't access the UI after disabling the setting?

blazoncek, 2022-11-11:
Yes. Get or self compile debug version and monitor serial output.
@srg74 provides debug binaries in his Wemos shield repository.

blazoncek, 2022-11-18:
If there are no updates I assume the issue is solved and will close in a few days.

shayded-exe, 2022-12-01:
I haven't had an opportunity to test with the debug binaries yet. Please keep open for now.

blazoncek, 2023-03-27:
4 months have passed without updates.
It looks like the issue is no longer relevant.
I am also unable to reproduce.

closed 2023-03-27 by blazoncek