wled/WLED #2880: Testing WLED v0.14.0-b0: Web UI not updating automatically after using JSON API to change state

cswaas, 2022-11-13:
### What happened?

I was previously using the official release v.0.13.3, where the web ui was automatically updating the visual state, when it was changed via the JSON API (i.e. turning on/off segment from Loxone controller with {"seg":[{"id":0, "on":"t"}]} ). The segment button was properly grayed out when turning off and highlighted white when turning on.

[image]

With v.0.14.0-b0, compiled on 13.11.2022, this automatic updating of the ui does not work anymore. When I reload the page, the state is updated correctly.

[image]

### To Reproduce Bug

Load v.0.14.0-b0 and follow the description above.

### Expected Behavior

The web ui should update automatically, when changing the state via the JSON API.

### Install Method

Self-Compiled

### What version of WLED?

WLED v0.14.0-b0 (2211111)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Using a QuinLED-Dig-Quad v3.1

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-11-13:
This change was intentional.
[image]

cswaas, 2022-11-13:
May I ask what the logic behind this new behavior is?
Shouldn‘t the ui always represent the actual state of the system?

Alternatively, can I use the JSON API in a way to have state changes represented in the WLED web ui?

blazoncek, 2022-11-13:
See commit comment.
https://github.com/Aircoookie/WLED/commit/0f6b1e4ae19ec58cdb2acd562e585b1fb7e2e1f4

If changing segment *on* state would cause `stateChange` that would also unload/deselect *current* preset which is not desired. We may need to find another way to broadcast such change over WS.

cswaas, 2022-11-13:
To be honest, I do not understand the new logic yet. 

When a preset is set "on" (i.e. also as visually activated button) and this preset has a certain segment also in the "on" state, then manually in the ui or via API turning "off" the certain segment should in my opinion turn "off" (deactivate) the preset as the system does not correspond to this preset. When activating the preset, the certain segment should turn "on" again. All this was the behavior in v.0.13.3, which made sense to me.

In v.0.14.0-b0 pushing the preset button does reload the preset state and turns "on" a previously turned "off" segment, but the preset button is not activated anymore. Yet another change in behavior that I do not understand.

I do not understand what you mean by "We may need to find another way to broadcast such change over WS." What is "WS"?

blazoncek, 2022-11-13:
WS == websockets, a method to inform clients of changes in WLED state. UI uses WS.
If you only need a response of the state after you call `{"seg":[{"id":0, "on":"t"}]}` add a `"v":true` property to the API call.
UI will be informed on next WS broadcast.

cswaas, 2022-11-13:
Here the ui in v.0.13.3:
-  1. Screenshot: Preset "Default" = "On" -Segment 0 = "On"
-  2. Screenshot: Preset "New Default" = "On" -> Segment 0 = "Off"

1. Screenshot
[image]

2. Screenshot
[image]

Here the ui in v.0.14.0-b0:

1. Screenshot
[image]

2. Screenshot
[image]

blazoncek, 2022-11-13:
I have noted your request/issue.
For now just change the mentioned line in json.cpp

cswaas, 2022-11-13:
Ok, thanks. I will try that.

cswaas, 2022-11-13:
I was able to apply your suggested change and it works as expected. Thanks again.

[image]

Aircoookie, 2022-11-13:
IMO it wouldn't be a problem if toggling segments changed state (aka. unloads preset and causes broadcast).
Changing power option should probably change state, only selection shouldn't.
Or was there a good reason for it that I forgot?

blazoncek, 2022-11-13:
@Aircoookie I am updating code ATM to reflect above discussion (along with other fixes).

closed 2022-11-13 by blazoncek