wled/WLED #2884: Reboot after adding 4th Segment on Wemos D1 Mini

cbcalhoun, 2022-11-15:
### What happened?

After adding the 4th segment on Genuine D1 Mini, Wemos begins crashing and rebooting. I am adding Segments groups in 2 and a spacing of 4. Unit will reboot several times then eventually failback to preset saved.

### To Reproduce Bug

Build 4 presets with groups of 2 and a spacing of 4. 
[image]

### Expected Behavior

Build a 4 color strand with groups of 2 leds of the same color and effect.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.3 (build 2208222)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-11-15:
Please try 0.14 beta since 0.13 is no longer maintained and will be replaced soon.

cbcalhoun, 2022-11-15:
Ok I will do that this evening. Thank you.

cbcalhoun, 2022-11-17:
Can you point me to the direction for the .14 beta code? I can't seem to locate it.

blazoncek, 2022-11-17:
Main branch.

blazoncek, 2022-11-25:
If the original problem is resolved by 0.14 please close this issue.

cbcalhoun, 2022-11-25:
I have not had a moment to do the OTA upgrade. I will work on this hopefully this weekend.

cbcalhoun, 2022-12-06:
I had a moment to do some testing now I am getting a crash and reload after two segments are assembled. 
[image]

blazoncek, 2022-12-07:
Use a debug build and monitor serial. When reboot happens copy crash dump and use exception decoder.
Then post your findings.

BTW out of 10 ESP8266 devices I have installed with multiple segments (not just two) I do not have any reboots. Use esptool to erase flash and upload WLED via serial.

blazoncek, 2022-12-07:
You can also verify the amount of free heap. Use info page.

cbcalhoun, 2022-12-07:
On the debug 0.14 bin. I cant use GPIO 1 for my second channel but I was able to create up to 5 seg without disconnection. Also on the .13 bin I have about 19+kB heap.

[image]

[image]

--DEBUG INFO---
Runtime: 690302
Unix time: 1670448487,302
                         Free heap: 17728
Wifi state: 3
State time: 450193
NTP last sync: 999000000
Client IP: 10.0.254.24
Loops/sec: 3977
UM time[ms]: 0/1
Strip time[ms]: 0/5
Segments: 5 -> 360B
                   Modes: 4*187=748B
                                    Data: 4*187=748B
                                                    Map: 2*0=0B

blazoncek, 2022-12-07:
GPIO1 is reserved for debug output.
You will need to dig deeper for the cause you are experiencing as WELD normally exhibits no such problem on my ESP8266.

cbcalhoun, 2022-12-07:
Here is the model I am using to compare. 

[image]

[image]

blazoncek, 2022-12-07:
I am using the same make and model.

cbcalhoun, 2022-12-07:
I will do some more digging here is my settings for the LED strings. I tried a knockoff make of the D1 with the same results. 

[image]

[image]
[image]

blazoncek, 2022-12-07:
Try lowering led count

cbcalhoun, 2022-12-07:
I appreciate your time working with me on this. I have just tried 30 leds on 1 segment only. I am able to get to the 5th segment: 0 - 5, 5-10, 10-15, 15-20, crash on adding 20-25

[image]

[image]

cbcalhoun, 2022-12-08:
Ok after starting from the default config after installing the original bin files I have discovered once the device is tied to my wireless network the issue starts occurring. 

If I go straight in after flashing I can setup many segments. I did up to 9 without issue as shown in the photo below. Once I join the wireless with my PSK and set the static IP, I go in and setup the segments and the hang occurs on the 4th segment on 13.3 and the 3rd segment on .14 beta.

Could there be something in joining the network stack or the complexity of the PSK that is causing this?

Pre Wireless Join:

[image]

Post Wireless Join:

[image]

cbcalhoun, 2022-12-08:
Seems that using my wireless network with a complex key has something to do with it. I created a hotspot on my phone and joined my laptop and my test WLED D1 Mini to it. I used a simple PSK of 0u8120u812 to join the controller to. I was able to build up to 8 segments without issue shown below. 

My internal home network PSK comprises of special characters such as ! @ and #, I feel this might have something to do with it but not positive. Hope this might help pinpoint the issue.

[image]

blazoncek, 2022-12-08:
Without crash dumps you are chasing ghosts.

cbcalhoun, 2022-12-08:
Got the debug output to serial but I am not sure this is what is needed. 

[image]

blazoncek, 2022-12-08:
Now look for the crash. You may need to scroll back in the output window.
Crash dump is followed by WLED start sequence which is pretty obvious.

cbcalhoun, 2022-12-08:
I am not getting a crashdump in the output that I can see, maybe it isn't crashing. I have the log from start-up to issue below after assembling the segments.

---WLED 0.14.0-b0 2212050 INIT---
esp8266 2_7_4_7
heap 27792
PIN ALLOC: Pin 1 successfully allocated by 0x89 (137)
Registering usermods ...
Reading config
Reading settings from /wsec.json...
JSON buffer locked. (3)
JSON buffer released. (3)
JSON buffer locked. (1)
Reading settings from /cfg.json...
Removing all.
PIN ALLOC: Pin 2 successfully allocated by 0x82 (130)
Successfully inited strip 0 (len 30) with type 22 and pins 2,255 (itype 2)
                                                                          PIN ALLOC: Pin 0 successfully allocated by 0x85 (133)
Starting usermod config.
JSON buffer released. (1)
Initializing strip
Usermods setup
lastReconnectAttempt == 0
Access point disabled (init).
Connecting to CBCalhoun...

Connected! IP address: 10.0.254.26
Init STA interfaces
mDNS started
FileRead: /index.htm
Not-Found HTTP call:
URI: /skin.css
FileRead: /skin.css
JSON buffer locked. (17)
JSON buffer size: 7249 for request: 6
                                     JSON buffer released. (17)
JSON buffer locked. (19)
JSON buffer released. (19)
JSON buffer locked. (17)
JSON buffer size: 1596 for request: 5
                                     JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 5500 for request: 5
                                     JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 4929 for request: 5
                                     JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 2129 for request: 5
                                     JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 4449 for request: 5
                                     JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 3009 for request: 5
                                     JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 2289 for request: 5
                                     JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 4529 for request: 5
                                     JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 2369 for request: 5
                                     JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 4129 for request: 5
                                     JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 3969 for request: 5
                                     JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 1649 for request: 5
                                     JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 2929 for request: 5
                                     JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 3089 for request: 5
                                     JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 853 for request: 5
                                    JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 1987 for request: 3
                                     JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
WS client connected.
JSON buffer locked. (12)
JSON buffer size: 1988 for WS request (1066).
                                             Sending WS data to a single client.
JSON buffer released. (12)
---DEBUG INFO---
Runtime: 30000
Unix time: 30,000
                 Free heap: 19504
Wifi state: 3
State time: 0
NTP last sync: 999000000
Client IP: 10.0.254.26
Loops/sec: 3290
UM time[ms]: 0/1
Strip time[ms]: 0/10
Segments: 1 -> 72B
                  Modes: 4*187=748B
                                   Data: 4*187=748B
                                                   Map: 2*0=0B
                                                              JSON buffer locked. (11)
JSON buffer released. (11)
JSON buffer locked. (12)
JSON buffer size: 1988 for WS request (1073).
                                             Sending WS data to a single client.
JSON buffer released. (12)
Local time: 13:26
                 JSON buffer locked. (12)
JSON buffer size: 1988 for WS request (1074).
                                             Sending WS data to multiple clients.
JSON buffer released. (12)
JSON buffer locked. (11)
JSON buffer released. (11)
JSON buffer locked. (12)
JSON buffer size: 2482 for WS request (1353).
                                             Sending WS data to a single client.
JSON buffer released. (12)
JSON buffer locked. (12)
JSON buffer size: 2482 for WS request (1354).
                                             Sending WS data to multiple clients.
JSON buffer released. (12)
JSON buffer locked. (11)
JSON buffer released. (11)
JSON buffer locked. (12)
JSON buffer size: 2946 for WS request (1633).
                                             Sending WS data to a single client.
JSON buffer released. (12)
JSON buffer locked. (12)
JSON buffer size: 2946 for WS request (1634).
                                             Sending WS data to multiple clients.
JSON buffer released. (12)
WS client disconnected.
---DEBUG INFO---
Runtime: 60012
Unix time: 1670505993,012
                         Free heap: 14400
Wifi state: 3
State time: 0
NTP last sync: 999000000
Client IP: 10.0.254.26
Loops/sec: 3139
UM time[ms]: 0/1
Strip time[ms]: 0/2
Segments: 3 -> 216B
                   Modes: 4*187=748B
                                    Data: 4*187=748B
                                                    Map: 2*0=0B

blazoncek, 2022-12-08:
Then you have some other issues. 8266 can only handle 2 WS clients. If your browser makes more WS connections then this is the reason for your issues.
Unless you see crash there is little I can help.

hcker2000, 2022-12-08:
This seems to have something to do with special chars in the wifi password. If I use a password with no special chars the wemos d1 behaves as expected. As soon as I add in the special chars it starts acting up.

Not a full crash as there is nothing being thrown when debug is enabled.

The same effect also seems to happen on esp32's (lolin32 for me) if you use special chars in the segment names (left 1/2 for example). In this case it dosn't lose connection right away but takes a second and the connection comes back but the presets.json is messed up when it comes back up.

blazoncek, 2022-12-08:
It must be a WiFi driver then. Open an issue with Espressif.

As for segment names: Officially only ASCII characters are supported (up to 30), though you can use UTF8 if names are very short.

cbcalhoun, 2022-12-08:
I doubt I am going to get anywhere with Epressif because of my limited knowledge so I will likely have to just build another SSID for these devices with a simpler PSK for now.

blazoncek, 2022-12-08:
That would be prudent for any IoT device.

cbcalhoun, 2022-12-08:
Different project same issue with the special character limitation for others interested. https://github.com/sidoh/esp8266_milight_hub/issues/410

hcker2000, 2022-12-14:
So after more testing you can still ping the device and go to /edit but the main interface via web or app no longer works after saving more than 3 segments.

It also does not have this behavior when wled is acting as the access point. I'm also not seeing anything in the browser console or network tabs that look out of place.

cbcalhoun, 2022-12-18:
Also after testing with ESP32 board there is no issue like that found with the Wemos D1 connecting to my specific wireless network with the same PSK. Created 12 segments without issue. 
[image]

FlyingEagle, 2023-02-12:
I do have the same issue, reboot after segmenting.
I have 459 LED.

I can do everything if I do not segment at all.

After beginning of segmentioning, the 8266 reboots suddenly and random.

I have / need 6 segments.

My Wifi PSK is alphanummeric with no special chars.

The segmentation names are free of special chars I use "seg01" e.g.

blazoncek, 2023-02-12:
Please use debug build and monitor serial output.
Catch crash dump and post it here.
https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/%40Aircoookie/v0.14.0-b2/Debug_builds

FlyingEagle, 2023-02-16:
The Output is [here](https://gist.github.com/FlyingEagle/6a72dd8272159ac77e316c6794382804)

After some hours of working, the Home Assistant Entry says disabled. I can only put the device of with a long press on it. after this I can not get the device again to live instead of bootin it by cable off.

After this I can not log in via install.wled.me again, the com port is there, but nothing happens.

By pressing some times on "reset device" on the console windows and pressing enter, I got a fast disappearing message: Terminal disconnected: AbortError: signal is aborted without reason.

I try this:

```
Install WLED
Do you want to install WLED 0.14.0-b1/2212222? All existing data will be erased from your device.

```
This works again:
```
Installing WLED
 
14%

This will take a minute.
Keep this page visible to prevent slow down
```
But I can not connect to device.

I tried:
```
Install WLED
Do you want to install WLED 0.13.3/2208222? All existing data will be erased from your device.
```

After some tries I got access, I took the ESP out of the PCB and re-program it.

Now, I have access again.

On the ESP is nothing else connected as the WS2812 1st-Pixel wiith 3.3V out of the ESP. One Button. Both connected to the default GPIO named in the GUI of the ESP.

I have one of the ESP connected to only one led, and it works the hole day. From time to time I went to the web-gui and play around with the colors, everything works fine.

As long til I do segmentation on the hole strip, do some animations or what ever to it, control color and presets on/off via home assistant, went to the gui, adjust some settings and back. The ESP some times suddendly went to walhalla or where ever. It seemed that the config is gone.

I tried to enable/disable wlan sleep, I tried with and without segmention, with and without animation, I tried different ESP (Wemos-Style, Uno-Style) - ok, all were 8266. Every ESP shows the same behaviour.
The WLAN Key is free of special chars only digits and letters.

As long as it works, I love this stuff ...

I hope I could help you to help me :-)

blazoncek, 2023-02-16:
I can see one crash, 5 resets, and 2 times low memory condition.
Something is accessing your memory starved ESP too often. Try disabling/disconnecting HA.
Upgrade to ESP32 if possible.

FlyingEagle, 2023-02-16:
I want to use HA, thats the main reason for using WLED.

FlyingEagle, 2023-02-16:
[image]

This is my setup.

Where can I find some more informations about mapping? I want to have the yellow parts act as one.

closed 2024-04-23 by github-actions[bot]