wled/WLED #2886: Presets crashes on new install after updating api (0.14.0 beta / main)

ewoudwijma, 2022-11-15:
### What happened?

Presets crashes on new install after updating api

### To Reproduce Bug

1) Delete presets.json in /edit
2) Add a new preset use current state
3) Go to the new preset api text area and add something e.g. "ledmap":1,
4) Press Save

Then you get this

[image]

Checking contents of presets.json in /edit: only {

### Expected Behavior

No crash

### Install Method

Self-Compiled

### What version of WLED?

latest main (0.14.0)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-11-16:
The bug is due to missing `"0":{}` in presets.json (since the file was deleted).
presets.json with `"0":{}` object is created on boot if the file does not exist. If you delete presets.json while WLED is running and do not reboot, presets.json becomes corrupt (missing 1st element) as no `"0":{}`object is created.

There is another bug:
presets.json may not be created on boot if WLED is compiled without EEPROM support.

ewoudwijma, 2022-11-16:
The file was deleted after the bug occurred, the bug also takes place after an erase and install of WLED. 
Erasing the file was done to continue.

So reproduce the error by doing a fresh install and update the first created preset.

The bug is not occurring if you have 2 presets

blazoncek, 2022-11-16:
Please examine presets.json after save of 1st preset. "0":{} is missing which is the root cause.
I am readying a fix.

MoonModules, 2022-11-16:
Cool, let me know when it is ready ;-)

blazoncek, 2022-11-16:
c47d6cffa88fa5a4a98ead4565b0c44891e1e699

closed 2022-11-16 by blazoncek