wled/WLED #2888: Android Galaxy S22 Ultra

KnuckleUpFPV, 2022-11-17:
### What happened?

App reports a bug when trying to discover lights. It crashes the app a few times then the phone says the app has a bug. Works great on my old phone s10 ultra. Something broke with this newest version.

### To Reproduce Bug

Try to add lights via discover lights.

### Expected Behavior

List of available devices show up to connect to the lights. 

### Install Method

Binary from WLED.me

### What version of WLED?

1.0.3

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-11-17:
Please open issue at the appropriate place.
https://github.com/Aircoookie/WLED-App

closed 2022-11-17 by blazoncek

KnuckleUpFPV, 2022-11-17:
> Please open issue at the appropriate place. https://github.com/Aircoookie/WLED-App

My apologies. I didnt realize there was a seperate place for the app. Thank you.