wled/WLED #2898: Unable to flash ESP8266

iprak, 2022-11-22:
### What happened?

I successfully flashed WLED_0.13.3_ESP8266.bin on a NodeMCU device esptool.py instructions mentioned [here](https://kno.wled.ge/basics/install-binary/). The chip starts with blue LED on, spits out gibberish and does not start WLED-AP. I verified that the device is fine by flashing an alternate Espurna firmware.

### To Reproduce Bug

I flashed ESP8266 NodeMCU as following

```
C:\IP\Projects\ESP8266\esptool>python esptool.py --port COM11 write_flash 0x0 ..\WLED_0.13.3_ESP8266.bin
esptool.py v4.5-dev
Serial port COM11
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ec:fa:bc:5f:7e:cd
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x000b6fff...
Compressed 749456 bytes to 507153...
Wrote 749456 bytes (507153 compressed) at 0x00000000 in 44.8 seconds (effective 133.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

```

### Expected Behavior

Access point should be created. Not sure if something useful is emitted on serial.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
exlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
sdlܟ<�l�<

�l�
c|����r�#�c��go�dog���c8ćlsd;lx�o�
�
$
��
b
'�|�l��b��og�l��l`�ogl n;���g
cl`sۓn
#$`�;lć|�
l$�l��g�Ada
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-11-22:
Some variants of NodeMCU are known to have issues with WLED beyond version 0.8.x.
It has been guessed that they have a faulty WiFi implementation.
Please use another MCU.

blazoncek, 2022-11-22:
BTW, Ada in your output is a sign that WLED started successfulyl.

iprak, 2022-11-22:
I read about that issue of some chips being potentially faulty. I never see the access point getting created. I am seeing the same behavior on 3 different devices which were previously successfully working with Espurna.

blazoncek, 2022-11-22:
I am going to close this issue as WLED can do nothing to resolve the underlying problem.
Please use other ESP dev board brands. In my experience Lolin Wemos D1 mini is an excellent choice. My recommendation would also be ESP32 as it offers more functions and better performance.

closed 2022-11-22 by blazoncek

iprak, 2022-11-24:
My chip is actually a Wemos NodeMCU which should be the same as D1 .. just slightly bigger. 

I debugged a little bit into this. It seems as if the chip was getting overwhelmed. I only have LEDs so after recompiling with various options turned off, the AP started appearing but it was inconsistent. If I quit out of the main `loop`, then it definitely started consistently. I will look more into this but my needs were very limited so for now I just wrote my own firmware based on Espurna.

juhojama, 2023-03-07:
Just my two cents: I was unable to flash the NodeMCU V2ESP8266 EX via the browser installer (https://install.wled.me/) but was able to flash with the ESPtool and then adding wifi access information via browser installer using WLED_0.14.0-b1_ESP8266.bin and esptool.py v4.5.1