wled/WLED #2900: Unable to add more than 5 "Color Order Override" settings

francisreader, 2022-11-23:
### What happened?

in 0.13.3 on a Athom High Power controller.
Trying to add more than 5 Color Order Override settings (have six connected props in which each prop has a ws2811 element with different order)
UX only allows 5 overrides, even through commit https://github.com/Aircoookie/WLED/commit/b8e23b2d7ea0b71b49b8f188e0148fa8a15d9fc1 shows that 10 should be possible. 

### To Reproduce Bug

Go to LED & Hardware setup, scroll to Hardware setup -> Color Order Override. Click "+" 5 times. Only "-" icon is now visible. 

### Expected Behavior

Should allow up to 10 Color Order Override

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

francisreader, 2022-11-23:
Looks like the UX has max overrides set to 5 
var customStarts=false,startsDirty=[],maxCOOverrides=5;

Even through the UX app code allows up to 10

francisreader, 2022-11-23:
Looks like commit 5c39d8d12e9e82ca8c93c24a7ed4fc07cc7193a0 restricted it to 5. Is there a reason?

blazoncek, 2022-11-23:
Yes. Memory and CPU constraints on ESP8266.
This is by design.
If you wish compile with custom setting.

francisreader, 2022-11-23:
Understood. 
My first time bouncing through the code and commits.
Built FastLED and arduino code many times so will attempt and increase the limit in a custom image. 

Thanks and keep on keeping on..