wled/WLED #2903: Enhance the BME280 v2 usermod

srg74, 2022-11-24:
BME280 v2 usermod have some issues needs to be resolved.
1. Usermod is not working without connection to MQTT.
2. Most of the latest usermods have "Enable" check mark and it would be wonderful to follow this example.
I know it is hard to find original code developer but last modification was made by @albarlow , maybe there is a chance to take a look in code and make some changes.

blazoncek, 2022-11-24:
@albarlow, @betaphi and @xBelladonna were original contributors.
IMO the usermod *needs* update to be in line with recent development.

xBelladonna, 2022-11-28:
Hi there, thanks for the ping! I'll see what I can do to implement the features stated above. I'm a bit busy and dealing with life stuff but I'll do my best to contribute something within the week.

blazoncek, 2022-11-28:
FYI I did bring it us to speed partially. It is still very inefficient and wasteful on resources, though.

xBelladonna, 2022-11-30:
Wowzers the entire usermod framework has changed into something amazing and unrecognizable to me as it's been a couple of years since I contributed that usermod, and it's going to take me a very long time to get up to speed with just how that works let alone how much the usermod code is different. If everyone's waiting on me to update it we'll probably be waiting some good few weeks to even months since I'm about to start new job and things will be crazy.

blazoncek, 2022-11-30:
> Wowzers the entire usermod framework has changed into something amazing and unrecognizable to me ...

Just a few additions/enhancements in API nothing major.

martin072, 2022-12-27:
Are there any known issues with compiling this usermod on an ESP32 (Lolin S2)?

Compiling for the ESP8266 (D1 Mini) works fine,  but when I try to compile it for the ESP32, I get a ton of warnings and some errors.

```
wled00/../usermods/BME280_v2/usermod_bme280.h:255:70: error: call of overloaded 'String(float&, uint8_t&)' is ambiguous
             publishMqtt("humidity", String(humidity, HumidityDecimals).c_str());

wled00/../usermods/BME280_v2/usermod_bme280.h:260:76: error: call of overloaded 'String(float&, uint8_t&)' is ambiguous
             publishMqtt("heat_index", String(heatIndex, TemperatureDecimals).c_str());
            
/Users/martin/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:76:18: note: candidate: 'String::String(long long int, unsigned char)'
         explicit String(long long, unsigned char base = 10);
                  ^~~~~~
/Users/martin/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:75:18: note: candidate: 'String::String(double, unsigned int)'
         explicit String(double, unsigned int decimalPlaces = 2);
                  ^~~~~~
/Users/martin/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:74:18: note: candidate: 'String::String(float, unsigned int)'
         explicit String(float, unsigned int decimalPlaces = 2);                 
```

softhack007, 2022-12-27:
Maybe it's necessary to give C++ a hint for finding the right constructor..

Try to change `TemperatureDecimals` into  `unsigned(TemperatureDecimals)` in the publishMqtt calls - and tell us if it helps.

martin072, 2022-12-28:
Yep, that did the trick. Many thanks!

albarlow, 2022-12-28:
Hi All, sorry for the late reply - been snowed under at work and various...

I don't know who the original codeowner was for this usermod but I made some significant modifications to the code to make it Home-Assistant-discoverable and to take advantage of things like the usermod interface, info screen etc.  Prior to my revisions, the purpose of the usermod seemed solely to be to read the environmental parameters (and calculate others based on the paremeters read) and record them out to MQTT.  With that in mind, it didn't really occur to me that this meant that the usermod could now have value without MQTT.

That said, I've just had a skim through the code and it looks to me like it _should_ work without MQTT connected - everywhere that I can see it attempt to publish a value to MQTT, it is wrapped in an `if (WLED_MQTT_CONNECTED){...}`, which should cause it to just step over it if MQTT is disabled(?)...I've also just tried running with MQTT disabled and I can still read the values locally whilst not publishing them.  I only have an ESP32 active at the moment (my ESP8266 setup that also used this usermod is currently out of action) so perhaps that makes a difference?

As regards the Enable check mark, unless I misunderstand what you're saying (or someone's already added the toggle in), there is already an Enable toggle that can be checked in the usermod settings screen (which was something else that did not previously exist).

closed 2023-07-17 by srg74