wled/WLED #2904: Preset selecting default palette instead of selected colors

drewcovi, 2022-11-25:
### What happened?

Toggling between presets that use Twinklyfox and select colors for the holiday, in 0.14.0-b0 the Default color palette is selected.

### To Reproduce Bug

Create a preset with Color Gradient using custom colors  (with Twinklyfox in my case but may exist for other effects).
Create another preset with Color Gradient using separate custom colors  (with Twinklyfox in my case but may exist for other effects).
Toggle between both.

### Expected Behavior

Preset retains selected colors and uses them in the effect.

### Install Method

Self-Compiled

### What version of WLED?

0.14.0-b0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
n/a
```

### Anything else?

Might already be a known issue, but figured I'd log it just in case. Thanks for the amazing platform here!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-11-25:
Tested but cannot reproduce.
Please post JSON API of both presets.

drewcovi, 2022-11-25:
Ah! this was actually due to the missing ledmap (I actually wasn't able to *override* my preset). You fixed it 3 days ago https://github.com/Aircoookie/WLED/commit/713509527a2591ae93ad22fce28c251fa6365199

I'll just pull the latest. Thanks!

closed 2022-11-25 by drewcovi