wled/WLED #2906: Emoji Preset Quick Load Not Recognized

drewcovi, 2022-11-25:
### What happened?

Worked in 13.3 but in 14.0 QL labels with Emojis return from presets.json as unrecognized characters.

Sent value via sockets:
```json
{"ib":true,"sb":true,"sc":false,"psave":3,"n":"Halloween","ql":"🎃","v":true,"time":1669401909}	1669401883.5561786
```
Retrieved value:
```json
{"3":{"on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":414,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"col":[[230,43,255],[0,0,0],[255,145,0]],"fx":80,"sx":107,"ix":195,"pal":5,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}],"n":"Halloween","ql":"�"}  }
```

### To Reproduce Bug

Attempt to save a preset quick load label as an emoji

### Expected Behavior

emojis are returned from presets.json

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-11-25:
Non ASCII characters are not supported.

Aircoookie, 2022-11-25:
They are supposed to be (and for me emoji QL seems to still work in 0.14). Will look into it.

blazoncek, 2022-11-25:
None of C string is by default Unicode compliant.
We'd need to use unicode functions and have much larger string buffers everywhere.

Aircoookie, 2022-11-25:
Probably using 8 bytes for QL buffer would already fix it :)

drewcovi, 2022-11-25:
Makes for a very compact and useful quick launch icon. Thanks for looking into it. Really love this work.

blazoncek, 2022-11-25:
A single unicode character can use as much as 4 bytes.

drewcovi, 2022-11-25:
Might have been obvious but to clarify, many of my existing quickload labels work great (with emoji) they're just getting stripped out on save (but I can obviously revert to 13.3 to get it back into my presets.json

Aircoookie, 2022-11-26:
Increasing buffer did the trick. This problem (unknown/corrupted character) may still occur e.g. in namelabel, however I'm not going to increase the buffer as 32b is already quite generous, and still enough for 10 CJK characters or 8 emoji, so supporting more than that is not worth the extra memory use.

closed 2022-11-26 by Aircoookie

drewcovi, 2022-11-26:
Interesting stuff! Thanks for the context and the fix! Based on what you mentioned, I could probably take a look at some of the js apis to real-time validate the input on the front end. But probably not totally needed as that's a **lot** of bytes. I'd be surprised if folks routinely hit that limit.

jezmck, 2023-05-09:
I found that I can use some, but not all emoji.

e.g. 👑 works, but not 🇪🇺 .

[image]

blazoncek, 2023-05-09:
Quick load can only accept 2 characters, ASCII characters.
Unicode characters may take much more than 3 bytes for one character.

commit 713bf66a (2022-11-26): Increase QL buffer to allow unicode characters

Fixes #2906