wled/WLED #2909: WLED crashes whenever you do anything on any version after 0.12.0

The-Techy, 2022-11-26:
### What happened?

After installing a number of WLED version from 0.12.0 up, I simply cannot control the leds. I've tested on a couple ESP8266 boards with the same issue. AP opens, I connect it to wifi, can access it, can change number of leds etc. Then, as soon as I do anything, e.g. pick a colour or effect, the little dot in the bottom left corner goes red, it disconnects then reconnect a couple seconds later. The leds don't change. Happens every time, consistently, with different versions, methods of uploading/compiling, different boards, even when nothing is connected except usb. Really weird.

### To Reproduce Bug

1.Install WLED (anything above 0.012.0), using any method (web installer, 3rd party web installer, compiling and upload through VSCode etc)
2. Open AP, enter wifi details
3. Find IP and connect
4. Try and change anything

### Expected Behavior

Should change leds, just like it does on 0.12.0

### Install Method

Binary from WLED.me

### What version of WLED?

All above 0.12.0

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
Note, this did not appear consistently, this was on around the 4th crash that it showed up

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
 
Exception (3):
epc1=0x40244282 epc2=0x00000000 epc3=0x00000000 excvaddr=0x400b7d61 depc=0x00000000
 
>>>stack>>>
 
ctx: sys
sp: 3fffec40 end: 3fffffb0 offset: 0190
3fffedd0:  3ffecbc1 ffff7b80 0000002a 4024441c  
3fffede0:  0000ff3f 4bc6a7f0 000148e7 3fff4e8c  
3fffedf0:  3fff4f2d 0000004f 3fff4c68 00000000  
3fffee00:  ffffffff 3fffc6fc 40101086 3fff4e8c  
3fffee10:  00000000 3fffee60 3fff6dcc 40244449  
3fffee20:  00000000 3fffee60 3fff6dcc 4024105e  
3fffee30:  3fff6c9c 3fff6ca0 00000006 00000001  
3fffee40:  4010383c 3fffc200 00000022 3fff6c9c  
3fffee50:  3fff4e8c 00000000 3fff6dcc 40241316  
3fffee60:  3fff4d74 3fff61bc 00000006 00000001  
3fffee70:  00000000 00000000 00000000 3fff4718  
3fffee80:  3fff6c9c 3fff4720 3fff471c 40286af8  
3fffee90:  0000003a 3b32a8c0 00000019 00000039  
3fffeea0:  00000000 00000019 3ffee250 4010130c  
3fffeeb0:  00000000 00000000 3ffecacf 3fff48f8  
3fffeec0:  3ffec9d6 3fff48f4 3fff4e8c 40284f31  
3fffeed0:  3fff4624 3fff69ec 3fff69ec 40100d0c  
3fffeee0:  00000008 3fff6624 3ffee250 3fff69ec  
3fffeef0:  3ffec9c8 00000000 3fff4e8c 40284399  
3fffef00:  0632a8c0 00000001 40105ae1 3ffee988  
3fffef10:  3ffedee0 401024c2 00000000 00000005  
3fffef20:  00000002 40102258 00000002 00000000  
3fffef30:  4010228d 00000002 00000000 3fff6a74  
3fffef40:  3fffdc80 00000000 3fff4e8c 4027a087  
3fffef50:  40000f49 3fffdab0 3fffdab0 40000f49  
3fffef60:  40000e19 00000005 000b7e74 00000000  
3fffef70:  00000000 aa55aa55 000000ed 40105c99  
3fffef80:  40105c9f 000b7e74 00000000 78f202d9  
3fffef90:  4010000d 3f798e95 66e4c054 6e85b772  
[... 250 lines cut ...]
3fffff40:  3fffdad0 3fff00f2 3fff2a48 3fff00f2  
3fffff50:  3fffdad0 3fff3c9c 40253ad3 3fffefa0  
3fffff60:  3fffdad0 3fff00f2 3fff2a48 4023558e  
3fffff70:  00000000 feefeffe 00efeffe feefeffe  
3fffff80:  00000000 00000000 00000001 3fff3c9c  
3fffff90:  3fffdad0 00000000 3fff3c5c 402360cc  
3fffffa0:  feefeffe feefeffe feefeffe 40253bc0  
<<<stack<<<
 
--------------- CUT HERE FOR EXCEPTION DECODER ---------------
 
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
 
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v3969889e
~ld
Ada
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ewoudwijma, 2022-11-29:
What environment settings are you using?

Not thoroughly tested but with these settings I am able to run it:

[env:esp8266_4MB_min]
extends = env:d1_mini
upload_speed = 460800 ;115200
build_flags = ${common.build_flags_esp8266}
  -D WLED_RELEASE_NAME=esp8266_4MB_min
  -D WLED_DISABLE_ALEXA
  -D WLED_DISABLE_BLYNK
  -D WLED_DISABLE_HUESYNC
  ; -D WLED_DISABLE_2D
  -UWLED_USE_MY_CONFIG

blazoncek, 2022-11-30:
IMO faulty flash or incorrect flash mode. Use DOUT flash mode.

The-Techy, 2022-11-30:
> What environment settings are you using?
> 
> Not thoroughly tested but with these settings I am able to run latest WLED:
> 
> [env:esp8266_4MB_min] extends = env:d1_mini upload_speed = 460800 ;115200 build_flags = ${common.build_flags_esp8266} -D WLED_RELEASE_NAME=esp8266_4MB_min -D WLED_DISABLE_ALEXA -D WLED_DISABLE_BLYNK -D WLED_DISABLE_HUESYNC ; -D WLED_DISABLE_2D -UWLED_USE_MY_CONFIG
> 
> Or try the compiled bin for this. See [here](https://github.com/srg74/WLED-wemos-shield/blob/master/resources/experimental/Firmware/AudioReactive/WLEDMM_0.14.0.8_esp8266_4MB_min.bin) (MM version)

I've tried using both the precompiled versions, and compiling my own (pretty much just select the enviroment as d1_mini and don't change anything else)

The-Techy, 2022-11-30:
> IMO faulty flash or incorrect flash mode. Use DOUT flash mode.

How do you change the flash mode? I didn't realise there were different ones

blazoncek, 2022-11-30:
using esptool (`-fm dout`).

The-Techy, 2022-12-14:
> using esptool (`-fm dout`).

Sorry for the late response, I've been a bit busy. I don't use esptool, but in vscode I added `board_build.flash_mode = dout` to the d1_mini env, which as I understand, should force it to use dout. I still get the same issue, it crashes a little slower now, but still crashes. Even tried updating it OTA with one of the precompiled bins, and still same issue. I've tried it on two boards now, both with the same issue. Could it be a faulty batch of boards?

The-Techy, 2022-12-14:
Just tried uploading it onto a board I got a couple months before the others, same seller, and it has the exact same issue. I therefore don't think it's an issue with the boards, since they all run WLED 0.12.0 perfectly. Also tried an ESP32 (again, same seller), which also ran 0.12.0 absolutely fine, and the identical issue happened.

TLDR: Tried almost every board I have (all from same seller, different purchase times), all with same crashing issue. They all worked fine on 0.12.0.

The-Techy, 2022-12-14:
Okay, just tried esptool with the exact same results. Even erased all the flash, didn't work. I can't even use AP mode, since it crashes there too. Really annoying.

blazoncek, 2022-12-14:
Perhaps consider getting a new ESP from another seller, diffrent brand, different model, etc.
8266 costs $2 or thereabouts.

The-Techy, 2022-12-15:
So, it appears as though I've been very stupid. I tried changing everything, cable, board, upload method, etc. EXCEPT, the device I was accessing the UI through! All this time I've been using my laptop to connect, but I tried my phone, and I can change the lights, no issue, without it crashing!

Now to figure out why my computer would disconnect every time. Any ideas?

The-Techy, 2022-12-15:
After some more testing, it's definitely just macOS that's the problem. Through bootcamp, on the same machine, wled works fine, it's only in macOS. Also, I've tried multiple browsers, and a wled desktop app, all with the same issue. What could be causing macOS to do this?

blazoncek, 2022-12-15:
Please close as this is apparently not a WLED issue after all.

closed 2022-12-15 by The-Techy