wled/WLED #2911: LED Outputs 9 & 10 light/drive 2x the number set in length

babblephish, 2022-11-27:
### What happened?

LED output 9 and 10 drive double the number set for led length. Set length to 1, 2 turn on, set 2, 4 turn on, etc. So, 2x the length is actually lighting up and being driven.

### To Reproduce Bug

Set up LED outputs 1-8 with any GPIO you want. you can set the length to 1. Then set LED output 9 and set length to 1, 2 pixels light up, set 2 then 4, set 5 then 10, etc. Set LED output 10 to a length of 1, 2 pixels light up and so on same as 9. GPIO for 9 & 10 do not matter, you can use anything available and get the same results.

### Expected Behavior

Only the length set should light up. 

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.3 and WLED 0.14.b0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-05-27:
Outputs 9 & 10 use I2S driver instead of RMT.
Please try 0.14.0-b2 (build greater than 2305000) as NeoPixelBus was updated to version 2.7.5.
Be sure to report back or close if the issue is gone.

babblephish, 2023-06-05:
This seems resolved with the updated library.

closed 2023-06-05 by babblephish