wled/WLED #2914: Sunrise/Sunset for all time-controlled presets

KanyonKris, 2022-11-29:
WLED is controlling holiday lights on my house. Currently I use a date range (ie Nov 26 - Dec 26 for Christmas) and on and off times. I'd like the on time to be 30 minutes after sunset each day.
Perhaps "Sunrise" and "Sunset" could be added to the Hour as a drop-down list and +/- minutes specify minutes after or before sunrise or sunset, see image below.

[image]

blazoncek, 2022-11-29:
Unfortunately no.
Too complex to implement currently.

dmig, 2024-05-27:
I'd sign for this for other reason: sunset-dusk interval is 15-60 (or more?) minutes, depending on season and latitude.

A dawn/dusk calculation, as I understand, differs just by 1 constant from sunrise/sunset. So, adding `dusk` and `dawn` options should be simple.