wled/WLED #2915: Wired button setting does not work in live mode

drhideg, 2022-11-29:
### What happened?

I have d1 mini with pushbutton configured as `Button 1` on `GPIO0` in interface. I set this because i like the internal brighness setting on long press. The short and double press configured over presets. The presets not handled while in live mode and the buttons looks like do not work as they should. I can make it work if i move https://github.com/Aircoookie/WLED/blob/98138a02e3feae68abd8ed571c3863978f131e50/wled00/wled.cpp#L104 before the conditionals https://github.com/Aircoookie/WLED/blob/98138a02e3feae68abd8ed571c3863978f131e50/wled00/wled.cpp#L84

### To Reproduce Bug

Wire a button and configure preset (ex-macro).

### Expected Behavior

Buttons must always work.

### Install Method

Self-Compiled

### What version of WLED?

v0.14.0-b0 "Hoshi" (build 2211261)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

Config:
```json
{
  "rev": [1, 0],
  "vid": 2211261,
  "id": {
    "mdns": "",
    "name": "mini1",
    "inv": "Light"
  },
  "nw": {
    "ins": [{
        "ssid": "MY-SSID",
        "pskl": 9,
        "ip": [0, 0, 0, 0],
        "gw": [0, 0, 0, 0],
        "sn": [255, 255, 255, 0]
      }
    ]
  },
  "ap": {
    "ssid": "WLED-AP",
    "pskl": 8,
    "chan": 1,
    "hide": 0,
    "behav": 0,
    "ip": [4, 3, 2, 1]
  },
  "wifi": {
    "sleep": true
  },
  "hw": {
    "led": {
      "total": 91,
      "maxpwr": 10000,
      "ledma": 60,
      "cct": true,
      "cr": false,
      "cb": 0,
      "fps": 30,
      "rgbwm": 3,
      "ld": false,
      "ins": [{
          "start": 0,
          "len": 91,
          "pin": [2],
          "order": 0,
          "rev": false,
          "skip": 0,
          "type": 22,
          "ref": false,
          "rgbwm": 0
        }
      ]
    },
    "com": [],
    "btn": {
      "max": 2,
      "pull": true,
      "ins": [{
          "type": 0,
          "pin": [-1],
          "macros": [0, 0, 0]
        }, {
          "type": 2,
          "pin": [0],
          "macros": [1, 0, 2]
        }
      ],
      "tt": 50,
      "mqtt": false
    },
    "ir": {
      "pin": -1,
      "type": 0,
      "sel": true
    },
    "relay": {
      "pin": -1,
      "rev": true
    },
    "baud": 1152,
    "if": {
      "i2c-pin": [-1, -1],
      "spi-pin": [-1, -1, -1]
    }
  },
  "light": {
    "scale-bri": 100,
    "pal-mode": 1,
    "aseg": true,
    "gc": {
      "bri": 1,
      "col": 2.8
    },
    "tr": {
      "mode": true,
      "dur": 10,
      "pal": 1
    },
    "nl": {
      "mode": 1,
      "dur": 5,
      "tbri": 0,
      "macro": 0
    }
  },
  "def": {
    "ps": 4,
    "on": true,
    "bri": 255
  },
  "if": {
    "sync": {
      "port0": 21324,
      "port1": 65506,
      "recv": {
        "bri": true,
        "col": true,
        "fx": true,
        "grp": 1,
        "seg": false,
        "sb": false
      },
      "send": {
        "dir": false,
        "btn": false,
        "va": false,
        "hue": true,
        "macro": false,
        "grp": 1,
        "ret": 0
      }
    },
    "nodes": {
      "list": true,
      "bcast": false
    },
    "live": {
      "en": true,
      "mso": false,
      "port": 5568,
      "mc": false,
      "dmx": {
        "uni": 1,
        "seqskip": false,
        "addr": 1,
        "mode": 4
      },
      "timeout": 25,
      "maxbri": false,
      "no-gc": true,
      "offset": 0
    },
    "va": {
      "alexa": false,
      "macros": [0, 0],
      "p": 0
    },
    "mqtt": {
      "en": false,
      "broker": "",
      "port": 1883,
      "user": "",
      "pskl": 0,
      "cid": "WLED-mini1",
      "topics": {
        "device": "wled/mini1",
        "group": "wled/all"
      }
    },
    "ntp": {
      "en": true,
      "host": "NTP-HOST",
      "tz": 2,
      "offset": 0,
      "ampm": false,
      "ln": 0,
      "lt": 0
    }
  },
  "ol": {
    "clock": 0,
    "cntdwn": false,
    "min": 0,
    "max": 29,
    "o12pix": 0,
    "o5m": false,
    "osec": false
  },
  "timers": {
    "cntdwn": {
      "goal": [20, 1, 1, 0, 0, 0],
      "macro": 0
    },
    "ins": [{
        "en": 1,
        "hour": 255,
        "min": 0,
        "macro": 0,
        "dow": 127
      }, {
        "en": 1,
        "hour": 255,
        "min": 0,
        "macro": 0,
        "dow": 127
      }
    ]
  },
  "ota": {
    "lock": false,
    "lock-wifi": false,
    "pskl": 7,
    "aota": false
  },
  "um": {}
}

```
Presets:
```json
{
  "0": {},
  "4": {
    "on": true,
    "lor": 1,
    "bri": 255,
    "transition": 10,
    "mainseg": 0,
    "seg": [{
        "id": 0,
        "start": 0,
        "stop": 91,
        "grp": 1,
        "spc": 0,
        "of": 0,
        "on": true,
        "frz": false,
        "bri": 255,
        "cct": 127,
        "col": [[0, 0, 0], [0, 0, 0], [0, 0, 0]],
        "fx": 87,
        "sx": 255,
        "ix": 128,
        "pal": 11,
        "c1": 255,
        "c2": 255,
        "c3": 31,
        "sel": true,
        "rev": false,
        "mi": false,
        "o1": false,
        "o2": false,
        "o3": false,
        "si": 0,
        "m12": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }
    ],
    "n": "4 Glitter"
  },
  "5": {
    "on": true,
    "lor": 1,
    "bri": 255,
    "transition": 10,
    "mainseg": 0,
    "seg": [{
        "id": 0,
        "start": 0,
        "stop": 91,
        "grp": 1,
        "spc": 0,
        "of": 0,
        "on": true,
        "frz": false,
        "bri": 255,
        "cct": 127,
        "col": [[0, 0, 0], [0, 0, 0], [0, 0, 0]],
        "fx": 38,
        "sx": 96,
        "ix": 255,
        "pal": 11,
        "c1": 255,
        "c2": 255,
        "c3": 31,
        "sel": true,
        "rev": false,
        "mi": true,
        "o1": false,
        "o2": false,
        "o3": false,
        "si": 0,
        "m12": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }
    ],
    "n": "5 Aurora"
  },
  "2": {
    "on": "t",
    "n": "2 |On/Off|"
  },
  "3": {
    "on": true,
    "lor": 0,
    "bri": 255,
    "transition": 10,
    "mainseg": 0,
    "seg": [{
        "id": 0,
        "start": 0,
        "stop": 91,
        "grp": 1,
        "spc": 0,
        "of": 0,
        "on": true,
        "frz": false,
        "bri": 255,
        "cct": 127,
        "col": [[0, 43, 84], [0, 128, 255], [0, 0, 255]],
        "fx": 2,
        "sx": 64,
        "ix": 255,
        "pal": 2,
        "c1": 255,
        "c2": 255,
        "c3": 31,
        "sel": true,
        "rev": false,
        "mi": false,
        "o1": false,
        "o2": false,
        "o3": false,
        "si": 0,
        "m12": 1
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }, {
        "stop": 0
      }
    ],
    "n": "3 LedFx/Breathe"
  },
  "1": {
    "win": "LO=1&P1=3&P2=5&PL=~",
    "n": "1 |CYCLE|"
  }
}

```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-11-29:
This is intentional as presets have no function while in live mode.
Though...

If you select to apply live mode only on main segment then your observation is correct, preset should be applied.
The problem in such case is that preset may change segment set-up (change main segment, length, etc) which could break live mode functionality.

drhideg, 2022-11-29:
So, best i can do is to maintain a fork/local copy, and let the [wled.cpp](https://github.com/Aircoookie/WLED/blob/9f1a7a1c2087f4c38b019c007b821e97064174f2/wled00/wled.cpp) as is, but add a call for `handlePresets()` after each `applyPreset([...])` in [button.cpp](https://github.com/Aircoookie/WLED/blob/9f1a7a1c2087f4c38b019c007b821e97064174f2/wled00/button.cpp) ?

```diff
@@ -22,6 +22,8 @@@@ void shortPressAction(uint8_t b)
    }
  } else {
    applyPreset(macroButton[b], CALL_MODE_BUTTON_PRESET);
+   handlePresets();
+   yield();
  }

  // publish MQTT message
@@ -41,6 +43,8 @@@@ void longPressAction(uint8_t b)
    }
  } else {
    applyPreset(macroLongPress[b], CALL_MODE_BUTTON_PRESET);
+   handlePresets();
+   yield();
  }

  // publish MQTT message
@@ -60,6 +64,8 @@@@ void doublePressAction(uint8_t b)
    }
  } else {
    applyPreset(macroDoublePress[b], CALL_MODE_BUTTON_PRESET);
+   handlePresets();
+   yield();
  }

  // publish MQTT message
```