wled/WLED #2921: DDP connections dropping out

krisnoble, 2022-12-03:
### What happened?

I have three strips each with separate controller/power supply that I want to treat as one long strip. I have them set up as DDP segments from a main controller as described in the wiki. The main controller is an ESP32 (AZ-Delivery DevKitC NodeMCU CP2102) and the remotes are ESP8266s (genuine LOLIN D1 Mini v4). 

The connections work fine at first but then suffer dropouts, causing the segments to revert to standalone mode. Occasionally they will re-sync by themselves but I usually have to reboot either the main or remote to get them reconnected. The dropouts seem to be random as far as I can tell.

I did initially try this with all D1 Minis and had the same issue, so I upgraded the main controller to an ESP32 per the recommendation in the wiki, but it seems like the problem persists.

I've checked that the main controller is running the ESP32 version, all controllers are running 13.3 installed via install.wled.me. I have the main controller added to Home Assistant but not the remotes.

Any help much appreciated!

### To Reproduce Bug

Connect multiple instances via the DDP segment technique

### Expected Behavior

Segments stay in sync

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.3 (build 2208222)

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-12-03:
This is known issue with UDP buffer getting full.
Unfortunately not much can be done. Try adding Ethernet option to your master device.

krisnoble, 2022-12-05:
Ah that's a shame.. thanks for the info though! I assume the same thing explains why I've also been seeing occasional issues with the normal WLED sync mode?

Just to make sure I understand, you're suggesting connecting the main device via ethernet, the others can still be wifi? The setup I currently have isn't suited to a physical connection but I'm thinking that in theory I could have a controller connected directly to the router via ethernet, which then drives three remotes which are wifi. Would that be reasonable?

Based on this suggestion I'm assuming the issue is the outgoing UDP buffer on the main rather than an incoming buffer on the remote?

blazoncek, 2022-12-05:
> Ah that's a shame.. thanks for the info though! I assume the same thing explains why I've also been seeing occasional issues with the normal WLED sync mode?

No, that would mean your network has problems.

> Just to make sure I understand, you're suggesting connecting the main device via ethernet, the others can still be wifi?

Yes.

> Based on this suggestion I'm assuming the issue is the outgoing UDP buffer on the main rather than an incoming buffer on the remote?

Yes. Perhaps it has something to do with your particular WiFi network as my tests (using Ubiquiti UniFi wireless) did not show any packet dropout until number of LEDs was extremely high.

closed 2024-04-23 by github-actions[bot]