wled/WLED #2922: WLAN very slow to connect and dropping out

zpin, 2022-12-03:
### What happened?

WLAN seems to be very unreliable on my D1 mini with an ASUS router.

### To Reproduce Bug

I just installed the latest version of WLED and it took very long to connect at all. First it said WLAN connection failed then when I went away from the PC and back eventually it did connect.

### Expected Behavior

WLAN should connect within seconds not several minutes.

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.3, 0.13.1, 0.13.0-b7

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

I have worked with the D1 mini before with custom applications and had the same issue there, it seems to be related to ASUS routers somehow, at least that seems to be the common factor with other people having a similar issue.
The way I fixed it there was to use `WiFi.setPhyMode(WIFI_PHY_MODE_11G);`. I tried the same with WLED (changed `WIFI_PHY_MODE_11N` to `WIFI_PHY_MODE_11G` in `wled.cpp`) and it actually fixed the issue for me. Looking at other people having issues with WLED and ESP8266 devices, maybe that's the solution for them as well?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

zpin, 2022-12-03:
It looks like my "fix" was just random luck, it still seems to have issues connecting sometimes (my other apps with the D1 mini work consistently with the `WIFI_PHY_MODE_11G` setting). I'll try to find out more.

josebiro, 2022-12-05:
It might not be just you. I've been trying to find a problem with frequent wifi drops as well. Reboot helps for a bit. This has been happening since the update to 0.13, It seems transient until it's not and then remedied by powercycle. I'm afraid I don't have any other data at this point.

AMoo-Miki, 2022-12-27:
As an experiment, can you enable a hotspot on your phone ([Android](https://www.androidauthority.com/mobile-hotspot-setup-631280/), [iPhone](https://support.apple.com/en-us/HT204023)) and have WLED connect to that? Make sure any compatibility related checkbox is selected on your hotspot settings to have wireless g/n enabled.

If the ESP is able to connect to the hotspot quickly, you will know to look into the router settings. If it takes too long to connect to the hotspot too, I would recommend, flash a simple connect and log sketch to find the code that causes the delay.

FYI, I have seen a bunch of threads about D1 Mini clones having WiFi issues ([GH](https://github.com/esp8266/Arduino/issues/7382), [Reddit](https://www.reddit.com/r/esp8266/comments/i1swdh/wemos_d1_mini_pro_wifi_issue/)).

zpin, 2023-01-25:
I found out that it actually connects to the WLAN just fine, it just isn't always able to obtain an IP. I see the DHCP server offering an IP in the router logs but the D1 doesn't seem to pick that up. If I assign a static IP it seems to work fine, at least with the dev build.
None of my other devices have any issue with the DHCP server so that's very odd.

ssirkakriss, 2023-08-20:
I've encountered some thing similar with my compiles with WLED-0.14.0-b4. the esp8266 (wemos style) boards would have trouble connecting to wifi after setting it up in Wled's AP.  And the Wled phone app would also have troubles connecting to the esp8266 boards as well.  

I found that disabling esp-now for the build process allowed my devices to connect.
```
[env:custom_build]
extends = env:d1_mini
build_flags = ${env:d1_mini.build_flags} -D WLED_DISABLE_ESPNOW -D LEDPIN=2 -D RLYPIN=12 -D USERMOD_DALLASTEMPERATURE -D USERMOD_FOUR_LINE_DISPLAY
lib_deps = 
  ${env:d1_mini.lib_deps}
  OneWire@~2.3.5
  olikraus/U8g2 @ ^2.28.8
```

I've not built a firmware file for my devices since v13.3. I really wanted to try out the new 2d stuff. It's great stuff

tablatronix, 2023-09-30:
I can only run esp866 with sleep disabled

daduke, 2023-12-23:
I can confirm that adding `WIFI_PHY_MODE_11G` fixes 8266 wifi issues with Asus APs (I need that line for all my 8266 projects - I'm using plain ESP-12Es)

tablatronix, 2023-12-23:
I will try that, just updates unifi and having tons of connectivity issues with my 8266s

blazoncek, 2023-12-23:
I can confirm that ESP8266 *still works and connects* to a WiFi (b/g/n/ac) after changing phy mode to G standard.
FYI I did not have any issues with connectivity, though.

tablatronix, 2023-12-29:
There is a tasmota issue thats says the same thing, suggestion was disable n and I know band steering and 5ghz can have some issues as well

closed 2024-05-05 by github-actions[bot]

sci666-2023, 2024-05-06:
for me the same .... just with one device ..all other 3 works well ....  connection breaks .... device not reachable ..but in wlan bonitor from my router connection is excellent .... wtf ?!

tablatronix, 2024-05-07:
Sounds like power issue if its just the one

sci666-2023, 2024-05-11:
Nope. 30watt usb Power supply. Alternstively direct to pc usb high Power plug. Its Just 1 led for testing on the strip... So thats definitive not a point

blazoncek, 2024-05-11:
but it is. bad LDO on ESP

sci666-2023, 2024-05-11:
But it works fine if I bent and hold the plug upwards....