wled/WLED #2924: WS2812 flicker when using additional PWM LED (Wemos D1 mini)

jayzakk, 2022-12-04:
### What happened?

On my default setup (WS2812 on GPIO4), I added a single PWM White LED channel, connected to GPIO2. 
The LED strip starts to flicker as soon as PWM > 0%, worst at about 70%.
Swapping the two GPIOs worked as expected.

### To Reproduce Bug

Connect a WS2812 strip to any GPIO except GPIO2 (I tried all available ones of the D1 Mini).
Then add a PWM channel to whatever GPIO you want. As soon as you set PWM level to >0%, the strip will start to flicker.

The only WS2812 GPIO usable is GPIO2. It doesn't matter, which GPIO you use for PWM. You don't even need to connect anything to the PWM port. I also tried to rule out any kind of interferences, adding additional caps, and using my bench power supply, and different D1 Mini on a breadboard, without success.

### Expected Behavior

Adding a PWM channel should not affect a WS28xx LED strip.

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.3, 0.14.0b

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-12-04:
Swap GPIOs. GPIO4 is not recommended for digital LEDs.
Also check what other GPIO pins allow HW accelerated PWM.

jayzakk, 2022-12-04:
> Swap GPIOs. GPIO4 is not recommended for digital LEDs. Also check what other GPIO pins allow HW accelerated PWM.

Wow, thanks for the hint -> I wasn't aware the 8266 does not have hardware PWM.
So, additionally to the software PWM needed, the GPIO4 (and basically all except GPIO2?) are not able to output the ws2812 protocol by hardware but have to bitbang also?

This would explain my sighting, where the yellow (ws2812) signal gets paused by a blue (PWM) change:

[image]

Makes absolute sense now, when both is bitbanging. 
Is there some documentation about which ports SHOULD be used (for which kind of output), and which NOT? Didn't see on first look.

Thanks again!

Aircoookie, 2022-12-04:
Indeed, running bit-bang WS2812 is not ideal on its own due to the tight timing requirements, with additional PWM you are bound to run into timing issues.
See https://kno.wled.ge/basics/getting-started/#default-gpio-usage , GPIO 2, 1, and 3 support digital LED hardware-driving, in that order of preference (pin 3 uses the DMA method, which uses 4x the memory)

closed 2022-12-04 by Aircoookie

blazoncek, 2022-12-04:
I am using GPIO1 and GPIO3 with dev boards without any issues.
With properly designed controller board, of course.