wled/WLED #2929: Would it be possible to give DDP an across the board brightness adjuster?

mkaatman, 2022-12-06:
**Is your feature request related to a problem? Please describe.**
My DDP bulbs aren't bright enough

**Describe the solution you'd like**
I don't know if it would be feasible, but having either a global DDP adjustment or a per device DDP adjustment for brightness might be handy. Something where you could add X% of additional brightness in an attempt to make bulbs match LED strip output more closely.

**Describe alternatives you've considered**
Running a separate WLED instance for each.

blazoncek, 2022-12-07:
Use "force maximum brightness" in Sync settings and brightness multiplier in LED settings.

closed 2022-12-07 by blazoncek

mkaatman, 2022-12-07:
That's an improvement! Thanks.