wled/WLED #2930: Native ESP-IDF Support

filzek, 2022-12-07:
In the current changes of market from ARDUINO to professional SDK like ESP-IDF is possible to move the solution to a native SDK without arduino?

blazoncek, 2022-12-07:
IMO infeasible. Too much code to rewrite.
Better start from scratch with new project.

softhack007, 2022-12-07:
From my developer's perspective, ESP-IDF is a buch of python scripts, build around a set or core libraries that constantly change, and they are full of crazy bugs. 

Espressif seems to have a hard time keeping their IDF releases stable and consistent. So as an engineer, I really prefer to use what we have today. A "professional SDK" is not of much use when it only causes extra work, instead of making development more effective.

...But that's just my personal view.

closed 2022-12-17 by softhack007

Aircoookie, 2022-12-17:
Agreed on this being out of scope.

As a side note, Arduino support for ESP32 is based on ESP-IDF, and (while involved), it is totally possible to compile a custom ESP-IDF for use with Arduino platform (and thus WLED).
Furthermore, most ESP-IDF functions are fully available, as the Arduino support is basically just an abstraction layer on top of ESP-IDF to make code more portable and reduce the need to write low level interface code on your own.