wled/WLED #2931: Usermod settings SCL and SDA not showing platformio values

ewoudwijma, 2022-12-07:
### What happened?

Usermod settings do not show values for build flags HW_PIN_SCL and HW_PIN_SDA on new install. Show -1 instead:

[image]

### To Reproduce Bug

erase flash: pio run --target erase

Set in platformio environment xyz
  -D HW_PIN_SCL=22 -D HW_PIN_SDA=21

build xyz

go to settings / usermod 

### Expected Behavior

[image]

### Install Method

Self-Compiled

### What version of WLED?

0.14.0 latest

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ewoudwijma, 2022-12-07:
Same applies for 

[image]

(HW_PIN_DATASPI, HW_PIN_MISOSPI, HW_PIN_CLOCKSPI)

blazoncek, 2022-12-07:
They do, as a placeholder (remove -1 or any other input by deleting digits). This is intentional as to not allocate GPIO during boot.
HW_PIN_... macros are just informational which GPIO is reserved (if used) for hardware accelerated IO.

Not a bug but intentional behaviour.

ewoudwijma, 2022-12-07:
I saw the placeholder functionality.

I would say setting hw_pin_sda/scl (or not setting) determines if gpio should be allocated or not.
Might be confusing for end users as build flags normally are shown as default in the ui.

We are currently reviewing the sda/scl functionality. After review will let you know our advice.

blazoncek, 2022-12-07:
> I would say setting hw_pin_sda/scl (or not setting) determines if gpio should be allocated or not. Might be confusing for end users as build flags normally are shown as default in the ui.

Well, it is not intended in the way you think. Confusing or not these setting are meant for advanced users who know how are they compiling and for what.
If you want to pre-allocate I2C and/or SPI GPIO pins, do so by introducing new compile time options. These may have nothing in common with HW_PIN_...
 
> We are currently reviewing the sda/scl functionality. After review will let you know our advice.

I wish you luck. I stumbled to a roadblock when trying to use 2 I2C devices on the same HW pins. Looks like a bug in Espressif/Arduino implementation. Since I2C is not used for LEDs and global SPI may conflict if using APS102 they are secondary in priority so I dropped any further development.

ewoudwijma, 2022-12-08:
> Well, it is not intended in the way you think. Confusing or not these setting are meant for advanced users who know how are they compiling and for what.

Is it documented anywhere? Discord links also okay.
Any names of advanced users to check 

Not to criticize your approach but get a general understanding how it is supposed to be used in different scenarios, improve documentation if needed and also test together multiple i2c devices

ewoudwijma, 2022-12-08:
I2C SPI Scrapbook:

highest level declarations are in pins_arduino.h:

[image]

If no HW_PIN values set in platformio and above variables commented out we see where these vars are used due to errors:

SCK and MOSI

NeoPixelBus/src/internal/*DotStarGenericMethod.h

[image]

NeoPixelBus/src/internal/DotStarEsp32DmaSpiMethod.h

[image]

const.h

[image]

[image]

pinmanager.cpp

[image]

... to be continued ...

blazoncek, 2022-12-08:
Search in #dev-general for the "I2C" keyword on Discord.
There was no documentation but a lot of discussion. Happened somewhere in September.

The advanced users in question were @softhack007 and @Matoran ( #2740 )

As for documentation: If you want clarifications or complaint about missing information/documentation open different kind of issue. Or ask on Discord.

As noted above the implementation of shareable I2C and SPI failed and since it is not crucial for workings of WLED, further development stopped beyond the entry fields in usermod settings. Which work as intended.
As shareable pins are exclusively used by usermods (except SPI which in case of APA102 cannot be shared ATM) it is up to usermods to implement proper use of I2C and SPI buses. They do get the information about which GPIO ports are intended as HW accelerated via global variables. HW_PIN_... constants are informational and only meant for intended override of platform default values. You do not need to define them to have HW accelerated I2C or SPI.

I do admit that this information (how & why) needs to be added somewhere. Perhaps in WLED-docs by someone that has time and will to do that and has played with usermods that use them.

In current implementation you have global variables `i2c_sda`, `i2c_scl`, `spi_mosi`, `spi_miso` and `spi_sclk' which are filled at boot with values entered in usermod settings page. Their intention is to represent HW accelerated GPIO pins. If they are set to anything other than -1 the appropriate interface is initialised (assuming no other code did that prior) with `Wire.begin()` and `SPI.begin()`. In such case pins are also pre-allocated to prevent unintended use in other patrs of WLED. They do not do anything beyond that.
Usermods that use I2C or SPI and use the same GPIO as global pre-allocated ones do not need to call `Wire.begin()` or `SPI.begin()` as it has been done at the boot time.

softhack007, 2022-12-08:
Hi, thanks @blazoncek and @ewoudwijma, thanks this explains a lot how WLED behaves today.

Proposal: give me some time to digest & numbercrunch this in my head. I'll need liquid cooling (beer) for this, so can do it on the weekend. Then i'll make a proposal in discord (SR-to-WLED team), about how I2C sharing could/should behave (kind of one-line requirements) including user perspective. We discuss in discord, then I try to improve the current implementation.
Agreed?

softhack007, 2022-12-08:
SPI sharing seems to be a different beast:
- NeoPixelBus DotStar (and maybe other) use SPI pins
- SD_card used SPI
- There is vspi and hspi - sometimes need to use one but not the other
- SPI sharing works by using the same MISO/MOSI/CLK, but usually a dedicates "SS" (chip select) signal per device is needed.

So we could first try to improve I2C only, and then see what can be done for SPI.

blazoncek, 2023-02-05:
resolved by #2955

closed 2023-02-05 by blazoncek