wled/WLED #2932: Reboot with the new OS7.50 from the Fritzbox 7590

PatricckW, 2022-12-08:
### What happened?

The WLED restarted after 59 seconds, since the new software 7.50 is on the Fritzbox and it was rebooted once.
I have see this bug still on the Beta Versions from the Router, with the old OS7.29 I have no problems.

### To Reproduce Bug

Install the Version 7.50 and reboot the WLED, than it will restart after 59 Sec

### Expected Behavior

No reboots 

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.3 (build 2208222)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2022-12-08:
Actually WLED does not know what kind of router it is connecting to, as we use the standard Wifi driver from espressif.

To better understand what happens, we first need to see if you get error messages on serial interface. Please connect your device to a Computer (using USB serial), and tell us what kind of messages you get on serial monitor (for example, you can use Arduino IDE serial monitor).

PatricckW, 2022-12-08:
That is very difficult vor me, it is from Cod.m without USB interface, only Wifi.
I use a Fritzbox 7590 from AVM, one of the most used Routers for Privat in Germany
[image]
[image]

softhack007, 2022-12-08:
Just to check if the current version works better,

please install WLED 0.14 (beta 0) from one of these sites:
* https://wled-install.github.io
* or from https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/WLED_wemos_shield/v0.14.0-b0

PatricckW, 2022-12-08:
I try version 14 but it is the same Problem

softhack007, 2022-12-08:
ok, then best to try another ESP32.

PatricckW, 2022-12-08:
I wait for a new one yes.
I geht the problem as I try the beta Version from AVM for the Router, wenn I go back to the Release all work fine, no restarts

PatricckW, 2022-12-09:
I try it today with an Devolo Repeater, which I have converted as an access point with own WLAN SSID
No it work fine, no Reboots yet

SmartHomeExpert, 2022-12-10:
@PatricckW: Had the same reboots after updating to fritzos 7.50.
The only quick solution was to change the WLED WiFi Access Point Setup. Set „AP opens“ to „never“.

Seems like WLED needs some WiFi information which the new fritzos doesn’t brodcast anymore…

PatricckW, 2022-12-11:
Thank you, but "to never" means the AP will not appear again?

PatricckW, 2022-12-11:
I try a other Controller today, same problem (I use it with mesh)
It happen if you reboot it on the first time

Lollo79, 2022-12-11:
Ich hab das Problem mal AVM geschildert vielleicht haben die noch eine Idee.

PatricckW, 2022-12-11:
> Ich hab das Problem mal AVM geschildert vielleicht haben die noch eine Idee.

I try that as OS7.50 was on Beta State, but it don´t help much

Lollo79, 2022-12-11:
> > Ich hab das Problem mal AVM geschildert vielleicht haben die noch eine Idee.
> 
> I try that as OS7.50 was on Beta State, but it don´t help much

I have Release State OS7.50 on my FritzBox 
[image]

Lollo79, 2022-12-11:
> Actually WLED does not know what kind of router it is connecting to, as we use the standard Wifi driver from espressif.
> 
> To better understand what happens, we first need to see if you get error messages on serial interface. Please connect your device to a Computer (using USB serial), and tell us what kind of messages you get on serial monitor when the restart happens (you can use Arduino IDE serial monitor for this).

a Serial Monitor Quote 
`
[17:25:55]Rebooting...
[17:25:55]ets Jun  8 2016 00:22:57
[17:25:55]
[17:25:55]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[17:25:55]configsip: 0, SPIWP:0xee
[17:25:55]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[17:25:55]mode:DIO, clock div:1
[17:25:55]load:0x3fff0018,len:4
[17:25:55]load:0x3fff001c,len:1396
[17:25:55]load:0x40078000,len:11536
[17:25:55]load:0x40080400,len:6404
[17:25:55]entry 0x400806b4
[17:25:56]AS: Generic I2S Microphone - left channel only.
[17:25:56]Ada
[17:26:55]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[17:26:55]Core 0 register dump:
[17:26:55]PC      : 0x4016c718  PS      : 0x00060330  A0      : 0x8016d0fe  A1      : 0x3ffe2250  
[17:26:55]A2      : 0x00000000  A3      : 0x00000010  A4      : 0x000000a2  A5      : 0x3ffe27d0  
[17:26:55]A6      : 0x3ffe893d  A7      : 0x3ffe89df  A8      : 0x8016c232  A9      : 0x3ffe2230  
[17:26:55]A10     : 0x00000001  A11     : 0x3ffe893d  A12     : 0x000000a2  A13     : 0x00000001  
[17:26:55]A14     : 0x00060f23  A15     : 0x00000010  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
[17:26:55]EXCVADDR: 0x00000004  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  
[17:26:55]
[17:26:55]ELF file SHA256: 0000000000000000
[17:26:55]
[17:26:55]Backtrace: 0x4016c715:0x3ffe2250 0x4016d0fb:0x3ffe22d0 0x4016d15b:0x3ffe2300 0x4008c0ca:0x3ffe2330
[17:26:55]
[17:26:55]Rebooting...
[17:26:55]ets Jun  8 2016 00:22:57
[17:26:55]
[17:26:55]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[17:26:55]configsip: 0, SPIWP:0xee
[17:26:55]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[17:26:55]mode:DIO, clock div:1
[17:26:55]load:0x3fff0018,len:4
[17:26:55]load:0x3fff001c,len:1396
[17:26:55]load:0x40078000,len:11536
[17:26:55]load:0x40080400,len:6404
[17:26:55]entry 0x400806b4
[17:26:56]AS: Generic I2S Microphone - left channel only.
[17:26:56]Ada`

Lollo79, 2022-12-11:
> To better understand what happens, we first need to see if you get error messages on serial interface. Please connect your device to a Computer (using USB serial), and tell us what kind of messages [...]

Log: `Using 'COM3' as serial port.
Showing logs:
[17:37:03]ets Jun  8 2016 00:22:57
[17:37:03]
[17:37:03]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[17:37:03]configsip: 0, SPIWP:0xee
[17:37:03]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[17:37:03]mode:DIO, clock div:1
[17:37:03]load:0x3fff0018,len:4
[17:37:03]load:0x3fff001c,len:1396
[17:37:03]load:0x40078000,len:11536
[17:37:03]load:0x40080400,len:6404
[17:37:03]entry 0x400806b4
[17:37:04]AS: Generic I2S Microphone - left channel only.
[17:37:04]Ada
[17:37:56]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[17:37:56]Core 0 register dump:
[17:37:56]PC      : 0x4016c718  PS      : 0x00060930  A0      : 0x8016d0fe  A1      : 0x3ffe2210  
[17:37:56]A2      : 0x00000000  A3      : 0x00000010  A4      : 0x000000a2  A5      : 0x3ffe3e88  
[17:37:56]A6      : 0x3ffe8ab9  A7      : 0x3ffe8b5b  A8      : 0x8016c232  A9      : 0x3ffe21f0  
[17:37:56]A10     : 0x00000001  A11     : 0x3ffe8ab9  A12     : 0x000000a2  A13     : 0x00000001  
[17:37:56]A14     : 0x00060523  A15     : 0x00000010  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
[17:37:56]EXCVADDR: 0x00000004  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  
[17:37:56]
[17:37:56]ELF file SHA256: 0000000000000000
[17:37:56]
[17:37:56]Backtrace: 0x4016c715:0x3ffe2210 0x4016d0fb:0x3ffe2290 0x4016d15b:0x3ffe22c0 0x4008c0ca:0x3ffe22f0
[17:37:56]
[17:37:56]Rebooting...
[17:37:56]ets Jun  8 2016 00:22:57
[17:37:56]
[17:37:56]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[17:37:56]configsip: 0, SPIWP:0xee
[17:37:56]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[17:37:56]mode:DIO, clock div:1
[17:37:56]load:0x3fff0018,len:4
[17:37:56]load:0x3fff001c,len:1396
[17:37:56]load:0x40078000,len:11536
[17:37:56]load:0x40080400,len:6404
[17:37:56]entry 0x400806b4
[17:37:57]AS: Generic I2S Microphone - left channel only.
[17:37:57]Ada
[17:38:56]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[17:38:56]Core 0 register dump:
[17:38:56]PC      : 0x4016c718  PS      : 0x00060330  A0      : 0x8016d0fe  A1      : 0x3ffe21d0  
[17:38:56]A2      : 0x00000000  A3      : 0x00000010  A4      : 0x000000a2  A5      : 0x3ffe24dc  
[17:38:56]A6      : 0x3ffe8885  A7      : 0x3ffe8927  A8      : 0x8016c232  A9      : 0x3ffe21b0  
[17:38:56]A10     : 0x00000001  A11     : 0x3ffe8885  A12     : 0x000000a2  A13     : 0x00000001  
[17:38:56]A14     : 0x00060f23  A15     : 0x00000010  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
[17:38:56]EXCVADDR: 0x00000004  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  
[17:38:56]
[17:38:56]ELF file SHA256: 0000000000000000
[17:38:56]
[17:38:56]Backtrace: 0x4016c715:0x3ffe21d0 0x4016d0fb:0x3ffe2250 0x4016d15b:0x3ffe2280 0x4008c0ca:0x3ffe22b0
[17:38:56]
[17:38:56]Rebooting...
[17:38:56]ets Jun  8 2016 00:22:57
[17:38:56]
[17:38:56]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[17:38:56]configsip: 0, SPIWP:0xee
[17:38:56]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[17:38:56]mode:DIO, clock div:1
[17:38:56]load:0x3fff0018,len:4
[17:38:56]load:0x3fff001c,len:1396
[17:38:56]load:0x40078000,len:11536
[17:38:56]load:0x40080400,len:6404
[17:38:56]entry 0x400806b4
[17:38:57]AS: Generic I2S Microphone - left channel only.
[17:38:57]Ada
`

codmdu, 2022-12-12:
I tested it with the [WLAN Pixel Controller](https://shop.codm.de/automation/pixel/30/wlan-pixel-controller-wled) from cod.m and my FRITZ!Box 7590 (FRITZ!OS 7.50 and Firmware-Version 154.07.50).   
I don't have any disconnections with WLED Version 13.3

PatricckW, 2022-12-12:
> I tested it with the [WLAN Pixel Controller](https://shop.codm.de/automation/pixel/30/wlan-pixel-controller-wled) from cod.m and my FRITZ!Box 7590 (FRITZ!OS 7.50 and Firmware-Version 154.07.50).

Do you use Mesh on the Fritzbox?

codmdu, 2022-12-12:
> > I tested it with the [WLAN Pixel Controller](https://shop.codm.de/automation/pixel/30/wlan-pixel-controller-wled) from cod.m and my FRITZ!Box 7590 (FRITZ!OS 7.50 and Firmware-Version 154.07.50).
> 
> Do you use Mesh on the Fritzbox?

Yes, with an FRITZ!WLAN Repeater.

Lollo79, 2022-12-12:
I Have This one [ABC! WLED Controller / ESP8266](https://shop.myhome-control.de/ABC-WLED-Controller-V31-ESP8266/HW10002) and This [ESP32 Dev Kit C](https://www.az-delivery.de/collections/esp32/products/esp32-dev-kit-c-unverlotet) on both the Same Problem.

PatricckW, 2022-12-13:
> I have a Wemos MINI D1 ESP32. It‘s connected to a mesh with a Fritz 1200 AX Repeater. To change the Access Point setting did not solve the problem 100%. After ~1 day I had the reboots again. Since I deactivated and activated „never“ once more, it works again. Very weird…
IF you reboot it manuall, it starts with the reboot again

Holgi001, 2022-12-16:
I have the same problem with AP from UniFi.  After the last update last week.  before everything was ok.

Janideluxe, 2022-12-16:
same problem here also with UniFi Access Points. Fritz OS 7.50 on Fritzbox 7590 is running too, but wifi is dissabled.

Lollo79, 2022-12-16:
Have a old 7560 as WLAN-Bridge no Mesh own SSID, Configured and no Restarts.

PatricckW, 2022-12-16:
It happened after the first restart, do the 7560 still have os7.50?

Lollo79, 2022-12-16:
No aktual FW for 7560 is not higher than 7.29

PatricckW, 2022-12-16:
With the os7.29 is all fine, that is the problem, it start with OS7.50

Lollo79, 2022-12-16:
AVM is not aware of any guilt.

>  However, the cause of the restarts are errors in your ESP32 WLED controllers themselves. A device should never restart on its own.

PatricckW, 2023-01-05:
Something new with this issue?

Lollo79, 2023-01-06:
Nope, I'm waiting for the FritzOS 7.51 to try again, because I have an IoBroker Infrastructur with scripts for the WLED Controller.

JimlyTheReal, 2023-01-15:
Hello fellow sufferers, well now I'm here too. 
Yes ESP32 with Fritz!OS 7.50 - after 60s reboot.
Unfortunately, changing the AP setting didn't help me.
Actually I only have the problem with WLED and ESP32.
Less with WLED and ESP8266!
And not at all, no matter which ESP, with any other software.
There are no problems with the guest wifi either.
Now I am speechless...

JimlyTheReal, 2023-01-15:
But what I don't understand is it only affects ESP32 with WLED and Fritz!OS 7.50.
No ESP32 with Tasmota for example - is it really an AVM problem then?

softhack007, 2023-01-15:
The exception dump from [here](https://github.com/Aircoookie/WLED/issues/3029#issuecomment-1383118625)
points to mdns, maybe in combination with AP "always off". 

It seems to be a null pointer exception deep inside espressif core components, so maybe WLED cannot do much about that. ESP8266  needs different core libraries (platform framework), which explains why only esp32 seems to be affected.

So it could help to disable mDNS, by removing the mDNS name in wifi settings. Also setting AP mode to "when disconnected" may help.

softhack007, 2023-01-15:
> But what I don't understand is it only affects ESP32 with WLED and Fritz!OS 7.50.

We have only heard of these problems from users of AVM Fritz.box hardware. It seems that all other wifi routers (99%) do not cause such problems. So it might be something special that AVM does with the wifi network.

> No ESP32 with Tasmota for example - is it really an AVM problem then?

Tasmota does not use mDNS, and has a different architecture, different core libraries, etc. So does not make much sense -technicially- to simply compare the two firmware and conclude that WLED is buggy.

Holgi001, 2023-01-15:
I have a UniFi AP and the same problem

Holger

softhack007, 2023-01-15:
With FritzBox as the main DSL router, or completely without any wifi hardware from AVM?

JimlyTheReal, 2023-01-15:
@softhack007: okay I'm just trying to understand it, neither WLED nor AVM, rather the combination of espressif and mDNS with some routers

softhack007, 2023-01-15:
> @softhack007: okay I'm just trying to understand it, neither WLED nor AVM, rather the combination of espressif and mDNS with some routers

Yes, possibly. The exception decoder dump we have seen shows a crash inside mDNS. The root cause is still not clear - bug inside mDNS, or something completely different.

JimlyTheReal, 2023-01-15:
> Yes, possibly. The exception decoder dump we have seen shows a crash inside mDNS. The root cause is still not clear - bug inside mDNS, or something completely different.

Okay i removed the default entry from mDNS - and - you said it - it ist still working - no reboort anymore.... 👍 
[image]

Holgi001, 2023-01-15:
> With FritzBox as the main DSL router, or completely without any wifi hardware from AVM?

I only use the fritzbox for the internet.  no wifi.

PatricckW, 2023-01-15:
I try that too, the First time up more than a Minute, but now it reboots again all 60sec.

Take it back on my test Devolo AP and it works fine.

JimlyTheReal, 2023-01-18:
I opened a service call with AVM and am in contact with them via email.

JimlyTheReal, 2023-01-25:
Sorry, but AVM has closed the service call because no abnormalities can be seen in the diagnostic data. They refer to the developers of the WLED software, they should get the problem under control.

softhack007, 2023-01-25:
> developers of the WLED software, they should get the problem under control.

Nice response, meaning something like "no idea - not our problem - live with it" 🤷. Did AVM indicate where to look for potential differences? Handling of network protocols, timing of client disconnects/reconnects, 2.4Ghz band support, security (WPA2), etc? Because it must be related to something that the new AVM firmware does differently, as versions before 7.50 seem to work.

Honestly, I have no idea where to look for problems in WLED "to get them under control". Wifi support is coming from espressif drivers, its nothing that was created by the WLED team.

Most users have routers from other manufacturers, and 95% of them seems to not cause such problems.  So I tend to making a recommendation to users for avoiding AVM hardware, as we have no idea what causes the problem neither "how to get it under control". So my suggestion would be to *close this WLED problem report*, as there is not enough information available to reproduce the problem without AVM hardware. 

A workaround exists, as there are a lot of other routers to chose from that do not cause such problems.

softhack007, 2023-01-25:
Actually, if you don't mind to wipe your device before installing, there might be a last thing you could test.

In the MM fork, we have a buildenv for ESP32 and the new espressif framework ESP-IDF 4.4.1. This is still experimental, but i use it on a few "standard esp32" devices and it works ok for me.

If the problem comes from espressif drivers, there could be a chance that the new platform framework includes a solution.
 
* clone the moonModules source code from https://github.com/MoonModules/WLED
* run `esptool erase_flash`(commandline) to wipe your device
* build & upload [`env:esp32_4MB_V4_min`](https://github.com/MoonModules/WLED/blob/1b3952c8895a5b1b7b1edf119cb84ad132e75de7/platformio.ini#L915) in VSCode (platformIO)

JimlyTheReal, 2023-02-02:
> * [`esp32_4MB_V4_min`](https://github.com/MoonModules/WLED/blob/1b3952c8895a5b1b7b1edf119cb84ad132e75de7/platformio.ini#L915)

Hi @softhack007,

Sorry for my late reply, but I'm a bit busy at the moment.
I had just compiled the recommended version and uploaded it to the ERP32.
The thing now runs for over 20 minutes - it never did that before!
This could be the solution ;-)

codmpm, 2023-02-27:
Got feedback from one of my customers with the esp32 variant and it seems to work. Thank you @softhack007.
I also tried this version for the esp8266 veriant (`esp8266_4MB_S`) but that did not work at another customer. 

Is the `esp32_4MB_V4` a newer espressif IDF version than `esp8266_4MB_S`?

softhack007, 2023-02-27:
@codmdu Thank you for the feedback, very much appreciated 👍

> Is the esp32_4MB_V4 a newer espressif IDF version than esp8266_4MB_S?

Indeed, it is. ESP32 is based on ESP-IDF, while ESP8266 has a completely different OS/framework (=not ESP-IDF). MM "esp8266_4MB_S" is using the same 8266 platform as in the "vanilla AC" version. So in case there are also problems with 8266 and FritzOS, a different solutions needs to be found. 

MM "esp32_4MB_V4" is an experimental build using `framework = espressif@5.1.1` (ESP-IDF 4.4.1 / arduino-esp32 2.0.4), while the standard ESP32 build has `framework = espressif32@3.5.0` (ESP-IDF 3.3.5 / arduino-esp32 1.0.6.4). Unfortunately we cannot simply upgrade the framework, as there are many subtle differences, breaking changes, compatibility problems, and lots of new bugs in ESP-IDF 4.4.x (like random crashes when AsyncWebServer uses LittleFS) ... So I would not recommend to use "V4" version in customer projects - unless you have no other options left.

codmpm, 2023-02-27:
We still ship 0.13.3 AC. Currently only two of our customers had the OS 7.50 Problem... both using repeaters (IIRC). One is the issue starter @PatricckW.
Only ESP32 are sold since a year or something. So, if someone has the issue we can point to the MM-binary for now.

The one customers who has an older version of our controller - which is esp8266 - has the problem. 
I see that it's not trivial to simply compile with the newer IDF.
Maybe I just upgrade him to a newer Controller.

Any ideas on when the newer IDF will land in WLED AC?
Thank you.

PatricckW, 2023-02-27:
I try it wenn I get it from @codmpm I have no restarts now. 

Some other Problems come up as I try  restore a Backup from my Controller with13.3 and my Http do not realy work on the update one. 

So I change back to the Controler with 13.3

codmpm, 2023-03-06:
For these who want to test, here's the esp32_4MB_V4_S binary. Simply flash it though Manual OTA.

DeepflashX, 2023-06-26:
hello and good day,

is this fixed now in the latest b3 Build? I saw something about a core update to 5.2.0?

Aircoookie, 2023-06-27:
For the record, most my devices are connected to a FritzBox 6490 cable with firmware 7.51-106637 BETA and I do not experience any sort of connectivity issue. I don't recall exactly, but I think it ran 7.50 before, also without issue. Something is definitely fishy here, as the issue seems quite isolated. Perhaps it only affects certain models, like the 7xxx generation, or even certain settings. I even have enabled mesh and same SSID for both 2.4 and 5 GHz and it all works just as it should.

codmpm, 2023-06-27:
> For the record, most my devices are connected to a FritzBox 6490 cable with firmware 7.51-106637 BETA and I do not experience any sort of connectivity or reset issue. 

Dear @Aircoookie, thanks for your reply on this. 
All customers of mine only report the problem with 7.50 (non-beta) when also using Fritz Mesh Repeaters. I'm currently not a aware of anyone having the issues when not using a repeater.

softhack007, 2023-06-27:
> is this fixed now in the latest b3 Build? I saw something about a core update to 5.2.0?

Actually no news. The core update is only for -S3/-S2/-C3, while esp32 still use the same platform as before.

Actually we have an experimental build environment now for platform 5.2.0 on esp32: [esp32dev_V4_dio80](https://github.com/Aircoookie/WLED/blob/f442aad9620b93e4ae15bc9160e73f5d25eb6d15/platformio.ini#L418C6-L418C23). This build env is however not recommended for anything but just for developer cross-checking. It may work or not work for you. And it may have new subtle bugs - it seems that espressif is not really at _not_ breaking previously working functionality.

wled-install, 2023-08-12:
For information: the issue was reported by one person running SR WLED 0.13.3 with FritzBox 7590 / Firmware 7.56 / Mesh activated. No issue after downgrade to SR WLED 0.13.2

codmpm, 2023-10-10:
We are currently investigating the problem as far as we can. 

Could not reproduce with Fritbox on Fritz!Os 7.57 (6890) and Fritz Repeater 1750E with Mesh on 7.31.
I have ordered a newer Fritz Repeater to test this with current firmware on the repeater side.

In the meantime customers of us had succes with the current wled-mm Version: 
[wled-mm-0.14.0-esp32_4MB_V4_s-loxone-mqtt-20231008.bin.zip](https://github.com/Aircoookie/WLED/files/12857049/wled-mm-0.14.0-esp32_4MB_V4_s-loxone-mqtt-20231008.bin.zip)

codmpm, 2023-10-11:
I've created a google sheet to gather the combinations where the error occurs. Would be great if you can add yout Fritz!Box/Fritz!Repeater Model and Firmware so that we can try to hunt this down:

https://docs.google.com/spreadsheets/d/15Aguj5zU1SwqqeCfna8r94lrWJanPTU4HCkCzvmHt3w/edit?usp=sharing

fir3drag0n, 2023-10-28:
> We (@codm) are currently investigating the problem as far as we can.
> 
> Could not reproduce with WLED 0.13.3, Fritbox 6890 on Fritz!OS 7.57 and Fritz Repeater 1750E with Mesh on Fritz!OS 7.31. I have ordered a newer Fritz Repeater to test this with current firmware on the repeater side.
> 
> In the meantime customers of us had succes with the current wled-mm Version: [wled-mm-0.14.0-esp32_4MB_V4_s-loxone-mqtt-20231008.bin.zip](https://github.com/Aircoookie/WLED/files/12857049/wled-mm-0.14.0-esp32_4MB_V4_s-loxone-mqtt-20231008.bin.zip)

Is there also a version with hue sync?

codmpm, 2023-10-30:
@fir3drag0n: Check here on how to compile your own binary: https://kno.wled.ge/advanced/compiling-wled/

somebuddy87, 2023-11-11:
I am using an WT32-ETH01 ESP32 Ethernet Variant without WiFi activated. IT also reboots after 60 Seconds.

[89 more replies]