wled/WLED #2935: JS error when loading preset with max number of segments

ezcGman, 2022-12-08:
### What happened?

I've upgraded my v13.3 install to the latest master branch commit. After reboot, I got this error:
[image]

It only happened when loading a specific preset, other presets worked fine. I even added newly added properties von v13 to v14 to each segment (`c1, c2, c3, o1, o2, o3, si, m12`), but still happened.

After some trying, I could nail it down it only happens if the preset as the max number of segments (in my case 32 for an ESP32). If I remove the last segment, it work again. Adding it again makes the error come back.

Using the Chrome console, I could nail it down to this line that fails: https://github.com/Aircoookie/WLED/blob/2c8dbb94fcf55f2e73547a289cdc5f80517bb8f0/wled00/data/index.js#L809

### To Reproduce Bug

Have a preset with the max num of segments and load it :) 

### Expected Behavior

JS, pls don't crash ;)

### Install Method

Self-Compiled

### What version of WLED?

643f3007928e0022d0ca6f39ee4079c881f8979a

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Preset I'm using:
```
{
	"0": {},
	"1": {
		"n": "All In One",
		"on": true,
		"bri": 46,
		"transition": 7,
		"mainseg": 0,
		"seg": [{
			"id": 0,
			"start": 0,
			"stop": 2171,
			"grp": 1,
			"spc": 0,
			"of": 0,
			"on": true,
			"frz": false,
			"bri": 255,
			"cct": 127,
			"n": "Everything",
			"col": [
				[0, 0, 255],
				[0, 0, 0],
				[0, 0, 0]
			],
			"fx": 65,
			"sx": 93,
			"ix": 128,
			"pal": 0,
			"sel": false,
			"rev": false,
			"mi": false,
			"c1": 128,
			"c2": 128,
			"c3": 16,
[... 1082 lines cut ...]
				[0, 0, 0]
			],
			"fx": 0,
			"sx": 128,
			"ix": 128,
			"pal": 0,
			"sel": false,
			"rev": false,
			"mi": false,
			"c1": 128,
			"c2": 128,
			"c3": 16,
			"o1": false,
			"o2": false,
			"o3": false,
			"si": 0,
			"m12": 0
		}]
	}
}
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-12-09:
Please check if https://github.com/Aircoookie/WLED/tree/selall-bugfix solves it.

closed 2022-12-09 by ezcGman

blazoncek, 2022-12-09:
@Aircoookie can you check if you agree with change regarding "+ Add segment" button?

ezcGman, 2022-12-16:
@blazoncek Did this ever get merged into main? Or fixed in another way? Thx!

blazoncek, 2022-12-16:
Not yet. Waiting for @Aircoookie to approve.
There seems to be an issue on ESP8266. Not necessarily related.