wled/WLED #2937: Compile issue if WLED_DISABLE_2D is defined

FreezyExp, 2022-12-09:
### What happened?

Code is still attempting to use the removed getPixelColorXY 

### To Reproduce Bug

define WLED_DISABLE_2D 

### Expected Behavior

defining WLED_DISABLE_2D removes WLED_DISABLE_2D related code

### Install Method

Self-Compiled

### What version of WLED?

0.14

### Which microcontroller/board are you seeing the problem on?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

drhideg, 2022-12-09:
This happened with me too in linking stage with
<details><summary>nodemcuv2</summary>
Executing task: C:\Users\drhideg\.platformio\penv\Scripts\platformio.exe run --environment nodemcuv2 

Processing nodemcuv2 (board: nodemcuv2; platform: espressif8266@3.2.0; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
*** use existing my_config.h ***
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (3.2.0) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.20704.7 (2.7.4)
 - tool-esptool @ 1.413.0 (4.13)
 - tool-esptoolpy @ 1.30000.201119 (3.0.0)  
 - toolchain-xtensa @ 2.40802.200502 (4.8.2)
Converting wled00.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- FastLED @ 3.5.0
|   |-- EspSoftwareSerial @ 6.8.5
|   |-- SPI @ 1.0
|-- IRremoteESP8266 @ 2.8.2
|-- ESPAsyncWebServerAircoookie @ 2.0.7+sha.b9349dc
|   |-- ESPAsyncTCP @ 1.2.2
|   |   |-- ESP8266WiFi @ 1.0
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|   |-- LittleFS(esp8266) @ 0.1.0
|-- ESPAsyncTCP @ 1.2.2
|   |-- ESP8266WiFi @ 1.0
|-- ESPAsyncUDP @ 0.0.0-alpha+sha.697c75a025
|-- NeoPixelBus @ 2.6.9
|   |-- SPI @ 1.0
|-- Wire @ 1.0
|-- SD(esp8266) @ 2.0.0
|   |-- SDFS @ 0.1.0
|   |   |-- ESP8266SdFat @ 1.1.0
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|-- SPI @ 1.0
|-- EEPROM @ 1.0
|-- ArduinoOTA @ 1.0
|   |-- ESP8266WiFi @ 1.0
|   |-- ESP8266mDNS @ 1.2
|   |   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|   |-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|   |-- ESP8266WiFi @ 1.0
|-- ESP8266WiFi @ 1.0
|-- LittleFS(esp8266) @ 0.1.0
|-- ESP8266WebServer @ 1.0
|   |-- ESP8266WiFi @ 1.0
Building in release mode
Compiling .pio\build\nodemcuv2\libf39\FastLED\FastLED.cpp.o
Compiling .pio\build\nodemcuv2\libf39\FastLED\bitswap.cpp.o
Compiling .pio\build\nodemcuv2\libf39\FastLED\colorpalettes.cpp.o
Compiling .pio\build\nodemcuv2\libf39\FastLED\colorutils.cpp.o
Compiling .pio\build\nodemcuv2\libf39\FastLED\hsv2rgb.cpp.o
Compiling .pio\build\nodemcuv2\libf39\FastLED\lib8tion.cpp.o
Compiling .pio\build\nodemcuv2\libf39\FastLED\noise.cpp.o
Compiling .pio\build\nodemcuv2\libf39\FastLED\platforms.cpp.o
Compiling .pio\build\nodemcuv2\libf39\FastLED\platforms\esp\32\clockless_rmt_esp32.cpp.o
Compiling .pio\build\nodemcuv2\libf39\FastLED\power_mgt.cpp.o
Compiling .pio\build\nodemcuv2\libf39\FastLED\wiring.cpp.o
Compiling .pio\build\nodemcuv2\src\FX.cpp.o
Compiling .pio\build\nodemcuv2\src\FX_2Dfcn.cpp.o
[... 779 lines cut ...]
c:/users/drhideg/.platformio/packages/toolchain-xtensa@2.40802.200502/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: FX.cpp:(.text._Z14mode_fireworksv+0x1b6): undefined reference to `Segment::setPixelColorXY(int, int, unsigned int)'
c:/users/drhideg/.platformio/packages/toolchain-xtensa@2.40802.200502/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: FX.cpp:(.text._Z14mode_fireworksv+0x216): undefined reference to `Segment::setPixelColorXY(int, int, unsigned int)'
c:/users/drhideg/.platformio/packages/toolchain-xtensa@2.40802.200502/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: FX.cpp:(.text._Z14mode_fireworksv+0x2e6): undefined reference to `Segment::setPixelColorXY(int, int, unsigned int)'
c:/users/drhideg/.platformio/packages/toolchain-xtensa@2.40802.200502/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\d1_mini\src\FX.cpp.o:(.text._Z9mode_rainv+0x0): undefined 
reference to `Segment::move(unsigned char, unsigned char)'
c:/users/drhideg/.platformio/packages/toolchain-xtensa@2.40802.200502/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\d1_mini\src\FX.cpp.o: in function `mode_rain()':
FX.cpp:(.text._Z9mode_rainv+0xb4): undefined reference to `Segment::getPixelColorXY(unsigned short, unsigned short)'
c:/users/drhideg/.platformio/packages/toolchain-xtensa@2.40802.200502/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: FX.cpp:(.text._Z9mode_rainv+0xd8): undefined reference to `Segment::move(unsigned char, unsigned char)'
c:/users/drhideg/.platformio/packages/toolchain-xtensa@2.40802.200502/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: FX.cpp:(.text._Z9mode_rainv+0xfb): undefined reference to `Segment::setPixelColorXY(int, int, unsigned int)'
c:/users/drhideg/.platformio/packages/toolchain-xtensa@2.40802.200502/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\d1_mini\src\FX.cpp.o: in function `mode_halloween_eyes()':FX.cpp:(.text._Z19mode_halloween_eyesv+0x1ba): undefined reference to `Segment::setPixelColorXY(int, int, unsigned int)'
c:/users/drhideg/.platformio/packages/toolchain-xtensa@2.40802.200502/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: FX.cpp:(.text._Z19mode_halloween_eyesv+0x1d7): undefined reference to `Segment::setPixelColorXY(int, int, unsigned int)'
[... cut ...] =========

Environment    Status    Duration
-------------  --------  ------------
d1_mini        FAILED    00:00:11.626
========================================================================================= 1 failed, 0 succeeded in 00:00:11.626 ========================================================================================= 

 *  The terminal process "C:\Users\drhideg\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'd1_mini'" terminated with exit code: 1. 
</details>

blazoncek, 2022-12-09:
I do not know which source you are compiling but [this](https://github.com/Aircoookie/WLED/blob/2e5b19575f234c0ea550dbe91e93172f523908e3/wled00/FX.h#L599) piece of code ensures the existence of 2D functions even if WLED_DISABLE_2D is defined.

I guess you may have caching issues.

drhideg, 2022-12-09:
I can reproduce it on a fresh Gitpod.

As I see linker need the missed functions `Segment::getPixelColorXY(unsigned short, unsigned short)`,  `Segment::setPixelColorXY(int, int, unsigned int)`, `Segment::move(unsigned char, unsigned char)` to make that evaluation work in runtime (1D or 2D) but those are not available because of the `WLED_DISABLE_2D` preprocessor directive.
https://github.com/Aircoookie/WLED/blob/2e5b19575f234c0ea550dbe91e93172f523908e3/wled00/FX.cpp#L1205

<details>
<summary>WLED - Gitpod Code</summary>
gitpod /workspace/WLED (main) $ pio run --environment d1_mini
Processing d1_mini (board: d1_mini; platform: espressif8266@3.2.0; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing fastled/FastLED @ 3.5.0
Unpacking  [####################################]  100%
Library Manager: FastLED@3.5.0 has been installed!
Library Manager: Installing git+https://github.com/Aircoookie/ESPAsyncWebServer.git @ ~2.0.7
git version 2.37.3
Cloning into '/home/gitpod/.platformio/.cache/tmp/pkg-installing-liux5w6e'...
remote: Enumerating objects: 53, done.
remote: Counting objects: 100% (53/53), done.
remote: Compressing objects: 100% (48/48), done.
remote: Total 53 (delta 2), reused 25 (delta 0), pack-reused 0
Receiving objects: 100% (53/53), 265.06 KiB | 1.37 MiB/s, done.
Resolving deltas: 100% (2/2), done.
Library Manager: ESPAsyncWebServerAircoookie@2.0.7+sha.b9349dc has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing ESPAsyncTCP
Unpacking  [####################################]  100%
Library Manager: ESPAsyncTCP@1.2.2 has been installed!
Library Manager: Installing makuna/NeoPixelBus @ 2.6.9
Unpacking  [####################################]  100%
Library Manager: NeoPixelBus@2.6.9 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing SPI
Verbose mode can be enabled via `-v, --verbose` option
*** use existing my_config.h ***
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (3.2.0) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 3.20704.7 (2.7.4) 
 - tool-esptool @ 1.413.0 (4.13) 
 - tool-esptoolpy @ 1.30000.201119 (3.0.0) 
 - toolchain-xtensa @ 2.40802.200502 (4.8.2)
Converting wled00.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Library Manager: Installing IRremoteESP8266 @ 2.8.2

Unpacking  [------------------------------------]    0%
Unpacking  [------------------------------------]    1%
Unpacking  [------------------------------------]    2%
Unpacking  [#-----------------------------------]    3%
Unpacking  [#-----------------------------------]    4%
Unpacking  [#-----------------------------------]    5%
Unpacking  [##----------------------------------]    5%
Unpacking  [##----------------------------------]    6%
Unpacking  [##----------------------------------]    7%
Unpacking  [##----------------------------------]    8%
Unpacking  [###---------------------------------]    8%
Unpacking  [###---------------------------------]    9%
Unpacking  [###---------------------------------]   10%
Unpacking  [###---------------------------------]   11%
Unpacking  [####--------------------------------]   11%
Unpacking  [####--------------------------------]   12%
Unpacking  [####--------------------------------]   13%
Unpacking  [#####-------------------------------]   14%
Unpacking  [#####-------------------------------]   15%
Unpacking  [#####-------------------------------]   16%
Unpacking  [######------------------------------]   16%
Unpacking  [######------------------------------]   17%
Unpacking  [######------------------------------]   18%
[... 538 lines cut ...]
FX.cpp:(.text._Z14mode_fireworksv+0xe6): undefined reference to `Segment::getPixelColorXY(unsigned short, unsigned short)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: FX.cpp:(.text._Z14mode_fireworksv+0x13e): undefined reference to `Segment::getPixelColorXY(unsigned short, unsigned short)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: FX.cpp:(.text._Z14mode_fireworksv+0x1b6): undefined reference to `Segment::setPixelColorXY(int, int, unsigned int)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: FX.cpp:(.text._Z14mode_fireworksv+0x216): undefined reference to `Segment::setPixelColorXY(int, int, unsigned int)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: FX.cpp:(.text._Z14mode_fireworksv+0x2e6): undefined reference to `Segment::setPixelColorXY(int, int, unsigned int)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/d1_mini/src/FX.cpp.o:(.text._Z9mode_rainv+0x0): undefined reference to `Segment::move(unsigned char, unsigned char)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/d1_mini/src/FX.cpp.o: in function `mode_rain()':
[... cut ...]
collect2: error: ld returned 1 exit status
*** [.pio/build/d1_mini/firmware.elf] Error 1
========================================================== [FAILED] Took 69.09 seconds ==========================================================

Environment    Status    Duration
-------------  --------  ------------
d1_mini        FAILED    00:01:09.090
===================================================== 1 failed, 0 succeeded in 00:01:09.090 =====================================================
</details>

blazoncek, 2022-12-09:
Segment class has definition (not just declaration) of all those functions in case of WLED_DISABLE_2D.
You can verify that in source (FX.h, linked above).
I am using mac and Windows machines to compile and none produces linker error.

softhack007, 2022-12-10:
I've just tried it on my PC, by adding -DWLED_DISABLE_2D to the build_flags in [env:nodemcuv2], and it works for me. 
```ini
[env:nodemcuv2]
...
build_flags = ${common.build_flags_esp8266} -D WLED_RELEASE_NAME=ESP8266  -DWLED_DISABLE_2D
````

HOWEVER when i put `#define WLED_DISABLE_2D` into `my_config.h`, then the build fails. Could this be the clue to explain build failures seen by @FreezyExp and @drhideg ? 

@FreezyExp where exactly did you define WLED_DISABLE_2D?

softhack007, 2022-12-19:
Update: i think this one is fixed.
I just put `#define WLED_DISABLE_2D` into my_config.h and compilation worked.

2D disabled:
```
RAM:   [======    ]  59.8% (used 48996 bytes from 81920 bytes)
Flash: [========  ]  84.4% (used 753772 bytes from 892912 bytes)
```
2D included:
```
RAM:   [======    ]  60.0% (used 49140 bytes from 81920 bytes)
Flash: [========= ]  89.2% (used 796460 bytes from 892912 bytes)
```

@FreezyExp and @drhideg please confirm if it works for you now, too.

blazoncek, 2022-12-20:
I did swap #includes.

drhideg, 2022-12-20:
Confirmed. Works as it should.
2D enabled:
```
RAM:   [======    ]  57.7% (used 47256 bytes from 81920 bytes)
Flash: [=======   ]  71.8% (used 749836 bytes from 1044464 bytes)
```

2D disabled:
```
RAM:   [======    ]  57.5% (used 47096 bytes from 81920 bytes)
Flash: [=======   ]  67.7% (used 707292 bytes from 1044464 bytes)
```

blazoncek, 2022-12-20:
Fixed in 0a3d9116025a34079827f6fc400eb94ac4f288c9

closed 2022-12-20 by blazoncek