wled/WLED #2941: WLED Data Transmission Error

paitchadze, 2022-12-10:
### What happened?

installed 3 times at least, deleted, installed again and etc. I'm unable to control LED's with an app. 
every time I press button to change color or pattern, this is what happens:
https://www.youtube.com/shorts/JstleMlPtY8

I use WS2812B 5m 144per meter led strip.

LED Preferences:

Maximum current 3000ma (tried even 5-9,000)
LED Voltage 5V default 55mA
GRB - 720
GPIO 4 (plugged in Pin 4 with jumper)

anything else? 

### To Reproduce Bug

Behaves like that on mine

### Expected Behavior

Should have changed the colors and etc

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2022-12-12:
This actually looks very much like you have wiring problems.
https://www.youtube.com/watch?v=Q9mLitVxF3c

You might need a level shifter, or maybe your wiring (gpio used) is not aligned with LEDs settings in WLED.

blazoncek, 2023-09-18:
No response received, closing.

closed 2023-09-18 by blazoncek